project-chip/connectedhomeip #29947: [Build] failed to build android CHIP Tool

BurievSardor, 2023-10-24:
### Build issue(s)

I installed latest command line tools
I installed Java 11

When I run the following the command:
# ./scripts/build/build_examples.py --target android-arm64-chip-tool build

I got the following error:
*************
Error: LinkageError occurred while loading main class com.android.sdklib.tool.sdkmanager.SdkManagerCli
2023-10-19 18:06:03 WARNING     java.lang.UnsupportedClassVersionError: com/android/sdklib/tool/sdkmanager/SdkManagerCli has been compiled by a more recent version of the Java Runtime (class file version 61.0), this version of the Java Runtime only recognizes class file versions up to 55.0
*************

sdkmanager is requiring Java 17, but script uses grade 7.1.1 which requires java 11.
How to solve this issue?

sanlingye, 2023-12-22:
Hi，Have you solved this problem？

Kartikgupta1502, 2024-08-14:
Hey, Was this resolved?

yunhanw-google, 2025-02-03:
we have upgraded the android build with java 17 and gradle 7.3.3, and kotlin 2.0.
https://github.com/project-chip/connectedhomeip/pull/33837, this issue should be resolved, feel free to reopen a new case if this is still problematic. thanks

closed 2025-02-03 by yunhanw-google