project-chip/connectedhomeip #29951: [CERT-TEST-FAILURE] Test_TC_OO_2_3 failure (websockets.exceptions.ConnectionClosedError)

interfer, 2023-10-24:
### Feature Area

Other

### Test Case

TC_OO_2_3

### Reproduction steps

Running the Test_TC_OO_2_3 on Telink LightingApp (Thread-commissioned) with
```
scripts/tests/yaml/chiptool.py tests Test_TC_OO_2_3 --nodeId $LIGHTING_NODE_ID
```
The **step 131** interrupts with an error, dropping websocket connection:
`websockets.exceptions.ConnectionClosedError: no close frame received or sent`
By default on the **previous step 130** the wait interval is set to 40sec. I've tried decreasing timeout 40->20sec and it finished like normally would. Also tried doing that in differrent setup and Thread network, identical behavior.
Logs are in attached file.

Also tried playing with `_KEEP_ALIVE_TIMEOUT_IN_SECONDS` value in `scripts/py_matter_yamltests/matter_yamltests/websocket_runner.py` , didn't help much

**Could you confirm the issue must be on my side or is it a common and known thing to other platforms as well?**

[Test_TC_OO_2_3.log](https://github.com/project-chip/connectedhomeip/files/13116936/Test_TC_OO_2_3.log)

### Bug prevalence

Each time I do this

### GitHub hash of the SDK that was being used

433e6769389c5e127e8d88418d1dae3150e39534

### Platform

python, other

interfer, 2023-11-08:
@Damian-Nordic Hi, we've tested it on the `nrf52840dk_nrf52840` and got the same result. Do you have any suggestions on that?

interfer, 2023-11-14:
@woody-apple @bzbarsky-apple

Figured out that variables `secs_since_valid_ping` and `secs_since_valid_hangup` in `connectedhomeip/third_party/libwebsockets/repo/lib/core/context.c` changed the behavior, test passed.
I've tried following values:
`	context->default_retry.secs_since_valid_ping = 360;`
`	context->default_retry.secs_since_valid_hangup = 370;`
 Just after ~300sec after test starts `chip-tool` doubles "LWS_CALLBACK_SERVER_WRITEABLE" log print and almost immediately closes connection and `LWS_CALLBACK_WSI_DESTROY` appears, and it's gone.
With the above changes I don't see "SERVER_WRITEABLE" twice and it works as in previous test steps.

I've met differrent values (ex. 400 for both) for these two in some other `libwebsocket` implementations but I believe there must be a proper way to fix that in Matter apart from my "hotfix" in third_party lib as submodule...

I would be happy for any advice on that, thanks in advance!

bzbarsky-apple, 2023-11-15:
@interfer Thank you for finding that!

It looks like the default values are 300 and 310s respectively.  The question is why there is no response to the ping, right?

But if we want to configure this in our code and we _think_ the timeout is on the server side we can try to do that in `examples/common/websocket-server/WebSocketServer.cpp` where it sets up the `lws_context_creation_info`.  In particular, we should be able to point its `retry_and_idle_policy` member to a `lws_retry_bo_t` that has the values we want, and see if that helps.

The problem is that we can't just set non-default values for the ping/hangup bits.  We have to also set `retry_ms_table`, `conceal_count`, and `jitter_percent`.  I guess we can copy those values from the defaults in `third_party/libwebsockets/repo/lib/core/context.c`, but seems pretty clunky.  :(

And fundamentally: this setup _sounds_ like it sends a ping and only times out if the ping is unanswered.  Why is the ping unanswered?

interfer, 2023-11-15:
@bzbarsky-apple Thanks for the response!

Not sure if it's 100% relevant but compared the Linux lighting-app (paired onnetwork) output in that scenario:
```
[1700044624.881205][98253:98253] CHIP:TOO: LWS_CALLBACK_SERVER_WRITEABLE
[1700044624.882424][98253:98253] CHIP:TOO: LWS_CALLBACK_WS_PEER_INITIATED_CLOSE
[1700044624.882494][98253:98253] CHIP:TOO: LWS_CALLBACK_CLOSED
[1700044624.882534][98253:98253] CHIP:TOO: LWS_CALLBACK_WSI_DESTROY
```
And normally (like when I decreased for ex. 40->20s  wait interval in test-step) Telink app would also work just like that.

But when it fails chip-tool would show this:
...* 130-th step of "Waiting for 40000ms" *
```
 [1699651222.917848][33047:33047] CHIP:TOO: LWS_CALLBACK_SERVER_WRITEABLE
 [1699651232.582908][33047:33047] CHIP:TOO: LWS_CALLBACK_SERVER_WRITEABLE
 [1699651242.582904][33047:33047] CHIP:TOO: LWS_CALLBACK_CLOSED
 [1699651242.582926][33047:33047] CHIP:TOO: LWS_CALLBACK_WSI_DESTROY
```
After 1st "LWS_CALLBACK_SERVER_WRITEABLE" in a few sec appears the duplicated and I assume the connection with initial one is lost and ping left unanswered therefore. And there's no message LWS_CALLBACK_WS_PEER_INITIATED_CLOSE also...

I added debug logs for WS to `scripts/tests/yaml/chiptool.py` and _the client tells that connection is closed_ (if I get it the right way) :
```
130. Running Wait 40000ms130. Running Wait 40000ms     
[PostProcessCheckType.IM_STATUS check] The test expects no error and no error occurred. 

***** Test Step 131 : 24c:Reads OnOff attribute from DUT 131. Running 24c:Reads OnOff attribute from DUT> TEXT 'json:{ "cluster": "onoff", "command": "read", "..._specifier": "on-off" }' [166 bytes]
DEBUG:websockets.client:> TEXT 'json:{ "cluster": "onoff", "command": "read", "..._specifier": "on-off" }' [166 bytes]
< PING '' [0 bytes]
DEBUG:websockets.client:< PING '' [0 bytes]
> PONG '' [0 bytes]DEBUG:websockets.client:> PONG '' [0 bytes]
% sending keepalive ping
DEBUG:websockets.client:% sending keepalive ping
> PING 69 5e 35 c7 [binary, 4 bytes]
DEBUG:websockets.client:> PING 69 5e 35 c7 [binary, 4 bytes]
= connection is CLOSED
DEBUG:websockets.client:= connection is CLOSED! failing connection with code 1006
DEBUG:websockets.client:! failing connection with code 1006
! failing connection with code 1006
DEBUG:websockets.client:! failing connection with code 1006
! failing connection with code 1006
DEBUG:websockets.client:! failing connection with code 1006
```

That's why I guess it's the server side and would like to hear the advice on how to make more functional fix here and less clumsy. Please, correct me if I go in the wrong direction :)

Should I try to implement the suggested above changes to `examples/common/websocket-server/WebSocketServer.cpp` with `lws_context_creation_info` ?

bzbarsky-apple, 2023-11-16:
If you can test those, that would be great, yes.  That should change the server behavior.

interfer, 2023-11-17:
@bzbarsky-apple Could You please take a look at my suggested changes in PR? 
I didn't change `retry_ms_table`, `conceal_count`, and `jitter_percent` as they are set by default in lib's `context.c` and app works fine now. Tested with `chip-tool` on Telink Lighting-app Thread commissioned.

Used this [issue](https://github.com/warmcat/libwebsockets/issues/2015) as a reference, some things became clear to me from there, may be useful to others next time.

closed 2023-11-17 by mergify[bot]

bzbarsky-apple, 2023-11-18:
> as they are set by default in lib's context.c

@interfer Where does that happen?  It looked to me like the values from your provided struct would be used, and if you're not initializing them chances are you get either garbage or 0 values, no?

interfer, 2023-11-20:
@bzbarsky-apple I replied in PR comment, they will be initialized from default_table