project-chip/connectedhomeip #29971: [1.3]  cpu usage is too high when i start chip-tool interactive server

jaggerwoo, 2023-10-25:
### Reproduction steps

my system：
``` shell
root@pi48g:~# uname -a
Linux pi48g 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

when i run `./chip-tool interactive server`, the cpu usage is nearly full
[image]

### Bug prevalence

nil

### GitHub hash of the SDK that was being used

nil

### Platform

raspi

### Type

Core SDK Performance Improvement

bzbarsky-apple, 2023-10-25:
Presumably this loop in `WebSocketServer::Run` (with all the time spent under `lws_service` per the profile):
```
    while (mRunning)
    {
        lws_service(context, -1);

        std::lock_guard<std::mutex> lock(gMutex);
        if (gMessageQueue.size())
        {
            lws_callback_on_writable(gWebSocketInstance);
        }
    }
```

bzbarsky-apple, 2023-10-25:
Note that I tried changing that value to other values, but it still seems like all the time is taking in polling under `lws_service`, so this may just be a limitation of libwebsockets...  Though I would have expected it to sleep until there is input.

wilbur-se, 2023-10-31:
I just do bellow changes to WebSocketServer, may improve it.

`index 46f9a63c9a..159649aada 100644
--- a/examples/common/websocket-server/WebSocketServer.cpp
+++ b/examples/common/websocket-server/WebSocketServer.cpp
@@ -95,7 +95,8 @@ void LogWebSocketCallbackReason(lws_callback_reasons reason)
         ChipLogDetail(chipTool, "LWS_CALLBACK_CLOSED");
         break;
     case LWS_CALLBACK_SERVER_WRITEABLE:
-        ChipLogDetail(chipTool, "LWS_CALLBACK_SERVER_WRITEABLE");
+        //dead loop print after call lws_callback_on_writable
+        //ChipLogDetail(chipTool, "LWS_CALLBACK_SERVER_WRITEABLE");
         break;

@@ -179,7 +181,7 @@ CHIP_ERROR WebSocketServer::Run(chip::Optional<uint16_t> port, WebSocketServerDe
 
     while (mRunning)
     {
-        lws_service(context, -1);
+        lws_service(context, 0);

@@ -205,4 +214,8 @@ void WebSocketServer::Send(const char * msg)
 {
     std::lock_guard<std::mutex> lock(gMutex);
     gMessageQueue.push_back(msg);
+    if (gWebSocketInstance)
+    {
+        lws_callback_on_writable(gWebSocketInstance);
+    }`