project-chip/connectedhomeip #29985: subprocess.CalledProcessError: Command '['ninja', '-C', '/home/patri/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.[BUG]

patrickwang0821, 2023-10-25:
### Reproduction steps

I use this link to setup Matter Device 
[https://community.arm.com/arm-community-blogs/b/internet-of-things-blog/posts/build-a-matter-home-automation-service-using-raspberry-pi-arm-virtual-hardware-and-python](url)

but i create the step at 8 & 9 always show
error: command '/usr/bin/aarch64-linux-gnu-gcc' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> grpcio-tools

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
ninja: build stopped: subcommand failed.
['ninja', '-C', '/home/patri/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']

Traceback (most recent call last):
  File "/home/patri/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 327, in install_packages
    subprocess.check_call(ninja_cmd, stdout=outs, stderr=outs)
  File "/usr/lib/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['ninja', '-C', '/home/patri/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/patri/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 795, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/patri/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 787, in main
    return EnvSetup(**vars(parse())).setup()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/patri/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 457, in setup
    result = step(spin)
             ^^^^^^^^^^
  File "/home/patri/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 606, in virtualenv
    if not virtualenv_setup.install(
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/patri/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 342, in install
    install_packages(gn_target)
  File "/home/patri/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 330, in install_packages
    raise subprocess.CalledProcessError(err.returncode, err.cmd,
subprocess.CalledProcessError: Command '['ninja', '-C', '/home/patri/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.

So How can i fix this problem

### Bug prevalence

everytimes

### GitHub hash of the SDK that was being used

db49235c39635582ea522929f9905af03e3114c7

### Platform

python, raspi

### Platform Version(s)

python 3.11.2 raspi Raspberry Pi OS Lite(64bit) 2023-10-10

andy31415, 2024-10-29:
The error seems to be not being able to install `grpcio-tools`.
Are you able to `pip install grpcio-tools` generally or in a virtualenv?

andy31415, 2024-10-29:
Platform also seems to state `python 3.11.2 raspi Raspberry Pi OS Lite(64bit)`. For our RPI compiles, we always use an ubuntu version. I would try with Ubuntu 24.04 to make sure you have the highest changes of success. Using RPiOS you are likely using something that not many (guessing nobody else?) are using.

closed 2024-10-29 by andy31415