project-chip/connectedhomeip #29986: [1.1] [FanControl] set FanMode to Auto, value of PercentSetting is 0xFF

mmt-10, 2023-10-25:
### Reproduction steps

1. set `FanMode` Attribute on `FanControl` Cluster to Auto(0x05)
2. 
`FanMode` was set to `Auto`, then [void MatterFanControlClusterServerAttributeChangedCallback(const app::ConcreteAttributePath & attributePath)](https://github.com/project-chip/connectedhomeip/blob/8f66f4215bc0708efc8cc73bda80620e67d8955f/src/app/clusters/fan-control-server/fan-control-server.cpp#L184) set `PercentSetting` to NULL.
However, 
`void MatterPostAttributeChangeCallback(const chip::app::ConcreteAttributePath & attributePath, uint8_t type, uint16_t size, uint8_t * value)`  `*value` is 0xFF.

How do I resolve it?

### Bug prevalence

always

### GitHub hash of the SDK that was being used

commit 8f66f4215bc0708efc8cc73bda80620e67d8955f (HEAD, tag: v1.1.0.1)

### Platform

core

### Type

Common Cluster Logic

bzbarsky-apple, 2023-10-25:
@mmt-10 The representation of spec "null" for an 8-bit unsigned integer/enum value in MatterPostAttributeChangeCallback is `0xFF`.

"null" does not mean "0"; it means "not a value".  The spec reserves some value space of this purpose, note: 0xFF is not a valid value for a nullable enum8 or uint8 or anything derived from them.

closed 2023-10-25 by bzbarsky-apple

mmt-10, 2023-10-25:
@bzbarsky-apple 
Thank you for your reply.
I understand it.

When I read `PercentSetting` after `FanMode` was set `Auto`, `PercentSetting` is `0`.
However, MatterPostAttributeChangeCallback is `0xFF`.
So, I confused.

bzbarsky-apple, 2023-10-26:
> When I read PercentSetting after FanMode was set Auto, PercentSetting is 0.

That is strange...  How are you reading it, exactly?