project-chip/connectedhomeip #29988: Could one fabric have many(not one) administrators?

whqxbs, 2023-10-25:
### Build issue(s)

Could one fabric have many(not one) administrators? For example, I pair one matter smart light with CHIP-TOOL, and then I get one fabric. Could I let another CHIP-TOOL join in that fabric?(As I know, if I use `OpenCommissioningWindow`, the two CHIP-TOOL will become two fabrics with the light.)

bzbarsky-apple, 2023-10-25:
A fabric can absolutely have multiple administrators, as long as they share key material.

Doing that with chip-tool specifically is pretty hard; it has no good way to provide key material externally short of manually messing with its storage files.

closed 2023-10-25 by bzbarsky-apple

whqxbs, 2023-10-25:
Could you tell me how to design an experiment instead of CHIP-TOOL to do the test, a fabric containing multiple administrators?

bzbarsky-apple, 2023-10-25:
You just need to make sure you have two nodes that have the same root cert and IPK, and operational certs signed by the same root for two different node ids...

How to do that depends on your specific PKI situation and where your certs/keys live and so on.