project-chip/connectedhomeip #30001: [Tooling] Tag for the TH SHA

cecille, 2023-10-25:
### Reproduction steps / Feature

We need a tag or branch for the TH SHA both so we have an official record of where this was taken and so devs can easily repro.

### Platform

core (please add to version below)

### Type

Manually tested with SDK

andy31415, 2023-10-25:
what is the SHA we need to tag? 19771ed7 or something else?

raju-apple, 2023-10-26:
Yeah , [19771ed7101321d68b87d05201d42d00adf5368f] is the SHA we made the final release for TH

raju-apple, 2023-11-10:
@andy31415 can we tag the above SHA ?

andy31415, 2023-11-15:
what should be the tag name? what is the description for it?

doublemis1, 2023-11-22:
I understand that the tag should be created for previous Matter releases Test Harness (Matter 1.0, Matter 1.1),

cecille, 2023-12-14:
@raju-apple - can you answer above about tag names?

raju-apple, 2023-12-15:
@andy31415 Lets do TH-Matter-1.2 ? ( Current Matter Test Harness-Fall 2023 version is based of this SDK SHA )

andy31415, 2024-01-04:
https://github.com/project-chip/connectedhomeip/releases/tag/TH-Matter-1.2 created

closed 2024-01-04 by andy31415