project-chip/connectedhomeip #30041: [BUG] Failed to reach the CommissioningStageComplete if network is unreachable during commissioning progress

lion6420, 2023-10-27:
### Reproduction steps

We use our own matter controller, and are testing our error handling function. Therefore, during the following step, the chip-all-cluster-app will be shutdown and network will be cut down purposely.

1. Controller (ubuntu PC) connect to raspberry-pi (chip-all-cluster-app) by ethernet
2. Pairing device by mdns discovery (example command: ./chip-tool pairing onnetwork 1 20202021)
3. During the commissioning progress, shutdown the chip-all-cluster-app and cut down the ethernet connection purposely
4. The commission progress will failed and the call back function OnCommissioningComplete is not called.

#### Commissioning without network cut down: [Log without network cutdown.txt](https://github.com/project-chip/connectedhomeip/files/13184169/Log.without.network.cutdown.txt)
```
[1698369771.645721][149159:149161] CHIP:CTL: Going from commissioning step 'SendPAICertificateRequest' with lastErr = 'src/app/CommandSender.cpp:263: CHIP Error 0x00000032: Timeout' -> 'Cleanup'
[1698369771.645731][149159:149161] CHIP:CTL: Performing next commissioning step 'Cleanup' with completion status = 'src/app/CommandSender.cpp:263: CHIP Error 0x00000032: Timeout'
[1698369771.645734][149159:149161] CHIP:CTL: Expiring failsafe on proxy 0x7fdd9c0021c0
[1698369771.645757][149159:149161] CHIP:DMG: ICR moving to [AddingComm]
[1698369771.645762][149159:149161] CHIP:DMG: ICR moving to [AddedComma]
[1698369771.645814][149159:149161] CHIP:EM: <<< [E:59831i S:44553 M:10708539] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
[1698369771.645830][149159:149161] CHIP:IN: (S) Sending msg 10708539 on secure session with LSID: 44553
[1698369771.646094][149159:149161] CHIP:DMG: ICR moving to [CommandSen]
[1698369771.646140][149159:149161] CHIP:DMG: ICR moving to [AwaitingDe]
[1698369772.002093][149159:149161] CHIP:EM: Retransmitting MessageCounter:10708539 on exchange 59831i Send Cnt 1
[1698369772.002208][149159:149161] CHIP:IN: (S) Sending msg 10708539 on secure session with LSID: 44553
[1698369772.387395][149159:149161] CHIP:EM: Retransmitting MessageCounter:10708539 on exchange 59831i Send Cnt 2
[1698369772.387451][149159:149161] CHIP:IN: (S) Sending msg 10708539 on secure session with LSID: 44553
[1698369772.962330][149159:149161] CHIP:EM: Retransmitting MessageCounter:10708539 on exchange 59831i Send Cnt 3
[1698369772.962434][149159:149161] CHIP:IN: (S) Sending msg 10708539 on secure session with LSID: 44553
[1698369773.887160][149159:149161] CHIP:EM: Retransmitting MessageCounter:10708539 on exchange 59831i Send Cnt 4
[1698369773.887228][149159:149161] CHIP:IN: (S) Sending msg 10708539 on secure session with LSID: 44553
[1698369775.571683][149159:149161] CHIP:EM: Failed to Send CHIP MessageCounter:10708539 on exchange 59831i sendCount: 4 max retries: 4
[1698369777.872659][149159:149161] CHIP:DMG: Time out! failed to receive invoke command response from Exchange: 59831i
[1698369777.872707][149159:149161] CHIP:CTL: Received failure response  when disarming failsafesrc/app/CommandSender.cpp:263: CHIP Error 0x00000032: Timeout
```

#### Network cut down case: [LogNetworkCutdownCase.txt](https://github.com/project-chip/connectedhomeip/files/13184171/LogNetworkCutdownCase.txt)
```
[1698370306.475366][149323:149325] CHIP:CTL: Failed to perform commissioning step 10
[1698370306.475372][149323:149325] CHIP:CTL: Going from commissioning step 'SendPAICertificateRequest' with lastErr = 'src/app/CommandSender.cpp:263: CHIP Error 0x00000032: Timeout' -> 'Cleanup'
[1698370306.475386][149323:149325] CHIP:CTL: Performing next commissioning step 'Cleanup' with completion status = 'src/app/CommandSender.cpp:263: CHIP Error 0x00000032: Timeout'
[1698370306.475392][149323:149325] CHIP:CTL: Expiring failsafe on proxy 0x7ff8d40021c0
[1698370306.475424][149323:149325] CHIP:DMG: ICR moving to [AddingComm]
[1698370306.475438][149323:149325] CHIP:DMG: ICR moving to [AddedComma]
[1698370306.475507][149323:149325] CHIP:EM: <<< [E:2653i S:49873 M:139577590] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
[1698370306.475523][149323:149325] CHIP:IN: (S) Sending msg 139577590 on secure session with LSID: 49873
[1698370306.475567][149323:149325] CHIP:DMG: ICR moving to [AwaitingDe]
```

Is it because the DisArmSafe command failed in DeviceCommissioner::CleanupCommissioning and without handling it?
```
else
    {
        // If we've failed somewhere in the early stages (or we don't have a failedStage specified), we need to start from the
        // beginning. However, because some of the commands can only be sent once per arm-failsafe, we also need to force a reset on
        // the failsafe so we can start fresh on the next attempt.
        GeneralCommissioning::Commands::ArmFailSafe::Type request;
        request.expiryLengthSeconds = 0; // Expire immediately.
        request.breadcrumb          = 0;
        ChipLogProgress(Controller, "Expiring failsafe on proxy %p", proxy);
        mDeviceBeingCommissioned = proxy;
        // We actually want to do the same thing on success or failure because we're already in a failure state
        SendCommand(proxy, request, OnDisarmFailsafe, OnDisarmFailsafeFailure,
                    /* timeout = */ NullOptional);
    }
```

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

e059202613e91da78884aec4e1041d76da0af9bf

### Platform

other

bzbarsky-apple, 2023-10-27:
> Is it because the DisArmSafe command failed in DeviceCommissioner::CleanupCommissioning and without handling it?

Yes, exactly.  Good catch.  https://github.com/project-chip/connectedhomeip/pull/30068 should fix this; would you be willing to verify that @lion6420?

lion6420, 2023-10-28:
Sure, I am willing to verify that. I can do the verification until I go back to the office on Monday @bzbarsky-apple .

I would like to ask about the ArmSafe. I am not quite understanding about this mechanism. Why can't the controller just delete the commssionee and make the commissioner back to the initial stage?

bzbarsky-apple, 2023-10-28:
@lion6420 Yes, nothing here is urgent enough to be worth worrying about over the weekend!

> Why can't the controller just delete the commssionee and make the commissioner back to the initial stage?

It can, in terms its own internal state.  Clearing the fail-safe is about putting the commissionee device (if it's reachable) into a state where the user can try again (perhaps with a different wireless network, if the issue was that the device could not join the IP network) immediately instead of having to wait for the fail-safe to time out.

lion6420, 2023-10-28:
@bzbarsky-apple Thanks for your reply, have a nice weekend!

lion6420, 2023-10-30:
Hi @bzbarsky-apple 
I upgraded the SDK to 02e7647bc6ae8c87012113fb69fcf18f865d075c (Fri Oct 27.), and verified the issue was fixed with your modification. 

Controller log: [HandleSendingErrorsDuringCommissioing.txt](https://github.com/project-chip/connectedhomeip/files/13201038/HandleSendingErrorsDuringCommissioing.txt)

bzbarsky-apple, 2023-10-31:
@lion6420 Thank you for checking that!

closed 2023-11-01 by mergify[bot]