project-chip/connectedhomeip #30089: [1.1] ble timeout problem

song7788, 2023-10-30:
### Reproduction steps

Hello develope:
     I have ported chiptool on the arm linux platform, and when using Bluetooth network distribution, timeout always occurs.
 My operation method is as follows: 
When using chiptool command to pair and unpair the same device in an environment with less interference, Bluetooth will always fail after multiple network distribution operations

log :
CHIP:BLE: BLE scan error: src/platform/Linux/bluez/ChipDeviceScanner.cpp:154: CHIP Error 0x00000032: Timeout
CHIP:BLE: Failed to stop discovery GDBus.Error:org.bluez.Error.InProgress: Operation already in progress

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

origin/v1.1-branch

### Platform

other

### Platform Version(s)

every time

### Type

Platform Issue

lion6420, 2023-11-01:
We encounter this issue also. The ble timeout is set to 10 sec  (both scan and connection are 10 sec) in `BLEManagerImpl`. In our test environmant, it always takes over 10 sec to scan the device (Maybe BLE signal is weak or messy..).

Therefore, we extend the timeout value to 60 sec and the issue is improved.

andy31415, 2024-10-29:
This seems to be a custom platform integration issue.
Marking as closed with the assumption that extending timeouts makes things better. Our Linux support is generally reasonably restricted for testing purposes only (i.e. right now specifically Ubuntu 24.04 on x64 and some support for RPI & Ubuntu 24.04 as well).

closed 2024-10-29 by andy31415