project-chip/connectedhomeip #30102: [BUG] Error 1 on GroupKeyManagement, write GroupKeyMap attribute

jonsmirl, 2023-10-30:
### Reproduction steps

Writing  GroupKeyManagement GroupKeyMap attribute. Doing read, add item, write to add groups one at a time.  On the fourth or fifth cycle this error occurs.

Error 1 is CHIP_ERROR_SENDING_BLOCKED but I think that error is bogus.

This doesn't look good. The WriteRequest is my attribute write.  It appears to be looping somehow when constructing the status response.

I (16496988) chip[EM]: >>> [E:23818r S:57283 M:19416322] (S) Msg RX from 2:AB0DE00000000000 [4507] --- Type 0001:06 (IM:WriteRequest)
I (16497008) chip[EVL]: Dropped 1 event from buffer with priority 1 and event number  0x000000000000005D due to overflow: event priority_level: 1
I (16497008) chip[EVL]: Dropped 1 event from buffer with priority 1 and event number  0x000000000000005E due to overflow: event priority_level: 1
I (16497018) chip[EVL]: Dropped 1 event from buffer with priority 1 and event number  0x000000000000005F due to overflow: event priority_level: 1
I (16497018) chip[EVL]: Dropped 1 event from buffer with priority 1 and event number  0x0000000000000060 due to overflow: event priority_level: 1
I (16497018) chip[EVL]: Dropped 1 event from buffer with priority 1 and event number  0x0000000000000061 due to overflow: event priority_level: 1
I (16497038) chip[EVL]: Dropped 1 event from buffer with priority 1 and event number  0x0000000000000062 due to overflow: event priority_level: 1
I (16497038) chip[EVL]: Dropped 1 event from buffer with priority 1 and event number  0x0000000000000063 due to overflow: event priority_level: 1
I (16497038) chip[EVL]: Dropped 1 event from buffer with priority 1 and event number  0x0000000000000064 due to overflow: event priority_level: 1
I (16497038) chip[EVL]: Dropped 1 event from buffer with priority 1 and event number  0x0000000000000065 due to overflow: event priority_level: 1
I (16497038) chip[EVL]: Dropped 1 event from buffer with priority 1 and event number  0x0000000000000066 due to overflow: event priority_level: 1
I (16497048) chip[EVL]: Dropped 1 event from buffer with priority 1 and event number  0x0000000000000067 due to overflow: event priority_level: 1
I (16497048) chip[EM]: <<< [E:23818r S:57283 M:44955213 (Ack:19416322)] (S) Msg TX to 2:AB0DE00000000000 [4507] [UDP:192.168.1.72%st1:49991] --- Type 0001:07 (IM:WriteResponse)

Why are all of these responses here?

2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D  WriteResponseMessage =
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D  {
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   AttributeStatusIBs =
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   [
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   AttributeStatusIB =
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   {
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   AttributePathIB =
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   {
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   Endpoint = 0x0,
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   Cluster = 0x3f,
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   Attribute = 0x0000_0000,
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   }
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D  
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   StatusIB =
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   {
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   status = 0x00,
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   },
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D  
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   },
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D  
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   AttributeStatusIB =
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   {
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   AttributePathIB =
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   {
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   Endpoint = 0x0,
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   Cluster = 0x3f,
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   Attribute = 0x0000_0000,
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   ListIndex = Null,
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   }
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D  
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   StatusIB =
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   {
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   status = 0x00,
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   },
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D  
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   },
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D  
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   AttributeStatusIB =
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   {
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   AttributePathIB =
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   {
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   Endpoint = 0x0,
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   Cluster = 0x3f,
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   Attribute = 0x0000_0000,
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   ListIndex = Null,
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   }
[... 59 lines cut ...]
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   ],
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D  
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D   InteractionModelRevision = 11
2023-10-29 22:38:58.534 10393-10516 DMG                     com.lowpan.m2                        D  }
2023-10-29 22:38

### Bug prevalence

always

### GitHub hash of the SDK that was being used

6bb0b716bc19c0e52d25a1812bc729ca202123b5

### Platform

core

### Anything else?

Not obvious how this works.
https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/group-key-mgmt-server/group-key-mgmt-server.cpp#L193

 VerifyOrReturnError(fabric_index == value.fabricIndex, CHIP_ERROR_INVALID_FABRIC_INDEX);
My client doesn't know its fabricId (which can vary by device), should't it ignore the fabricId I sent on the write?

jonsmirl, 2023-10-30:
The mess above should have been a constraint error since CHIP_CONFIG_MAX_GROUPS_PER_FABRIC is 4.

Groups per fabric is defaulted way too low. Group keys per fabric can be a low number like 4. However, Groups don't use much in resources. They should be defaulted significantly higher.  Looks like they are defaulted to 4 currently. My keypad need 12-15 groups. I will try increasing this in my Espressif build.

I do notice this which makes more sense. Espressif must not have this set.
controller/python/BUILD.gn:    defines += [ "CHIP_CONFIG_MAX_GROUPS_PER_FABRIC=50" ]

jonsmirl@ares:~/aosp/esp-matter/connectedhomeip/connectedhomeip/src$ grep -rI CHIP_CONFIG_MAX_GROUPS_PER_FABRIC *
controller/python/BUILD.gn:    defines += [ "CHIP_CONFIG_MAX_GROUPS_PER_FABRIC=50" ]
credentials/GroupDataProvider.h:    GroupDataProvider(uint16_t maxGroupsPerFabric    = CHIP_CONFIG_MAX_GROUPS_PER_FABRIC,
lib/core/CHIPConfig.h: * @def CHIP_CONFIG_MAX_GROUPS_PER_FABRIC
lib/core/CHIPConfig.h:#ifndef CHIP_CONFIG_MAX_GROUPS_PER_FABRIC
lib/core/CHIPConfig.h:#define CHIP_CONFIG_MAX_GROUPS_PER_FABRIC (4 * CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC)
lib/core/CHIPConfig.h:#if CHIP_CONFIG_MAX_GROUPS_PER_FABRIC < (4 * CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC)
lib/core/CHIPConfig.h:#error "Please ensure CHIP_CONFIG_MAX_GROUPS_PER_FABRIC meets minimum requirements. See Group Limits in the specification."
lib/core/CHIPConfig.h: * @def CHIP_CONFIG_MAX_GROUPS_PER_FABRIC
jonsmirl@ares:~/aosp/esp-matter/connectedhomeip/connectedhomeip/src$ grep -rI CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC *
lib/core/CHIPConfig.h: * @def CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC
lib/core/CHIPConfig.h:#ifndef CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC
lib/core/CHIPConfig.h:#define CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC 1
lib/core/CHIPConfig.h:#define CHIP_CONFIG_MAX_GROUPS_PER_FABRIC (4 * CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC)
lib/core/CHIPConfig.h:#if CHIP_CONFIG_MAX_GROUPS_PER_FABRIC < (4 * CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC)
jonsmirl@ares:~/aosp/esp-matter/connectedhomeip/connectedhomeip/src$

bzbarsky-apple, 2023-10-31:
Yeah, we are returning generic FAILURE, when arguably it should be CONSTRAINT_ERROR....  @lpbeliveau-silabs @jmartinez-silabs