project-chip/connectedhomeip #30106: [BUG] BLE Commissioning failing on ESP32-C6

kyberpunk, 2023-10-30:
### Reproduction steps

- Using ESP32-C6-DevKitM-1
- Build examples/lighting-app/esp32 for ESP32-C6 according to: https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/build_app_and_commission.md
```
idf.py --preview set-target esp32c6
idf.py -D 'SDKCONFIG_DEFAULTS=sdkconfig.defaults.esp32c6' build
```
- May need to change SPI flash to 4MB (in case of DevkitM)
- Run device and execute ble-wifi pairing with chiptool
- Following errors will appear on ESP32 device during commissioning:
```
I (338018) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 1) status= 14 
I (338068) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 1 12
I (338068) chip[EM]: >>> [E:5197r S:0 M:161869132] (U) Msg RX from 0:A3EE5FA15E5B387D [0000] --- Type 0000:40 (SecureChannel:StatusReport)
E (338088) chip[SC]: Received error (protocol code 2) during PASE process: 38
E (338098) chip[SC]: Failed during PASE session setup: 38
E (338098) chip[SVR]: Commissioning failed (attempt 1): 38
I (338108) chip[BLE]: Releasing end point's BLE connection back to application.
```
The same issue occurred with provisioning with Homeassistant or CHIPTool on Raspberry Pi 4 and CHIPTool on Android phone.
I would assume that problem will be on ESP32 device side. 

### Bug prevalence

Every commissioning attempt with BLE

### GitHub hash of the SDK that was being used

v1.2.0.1, v1.1.0.2

### Platform

esp32

### Platform Version(s)

esp-idf 5.1.1

### Anything else?

[log.txt](https://github.com/project-chip/connectedhomeip/files/13209721/log.txt)
[sdkconfig.txt](https://github.com/project-chip/connectedhomeip/files/13209731/sdkconfig.txt)

shripad621git, 2024-01-03:
@kyberpunk , the error `Received error (protocol code 2) during PASE process: 38` stands for `invalid PASE parameter.`Can you please verify that you are prvoviding the corerct discriminator and passcode while commissioning using chip-tool or scanning the correct qrcode for commissioning using other ecosystems.
Moreover, I have also tried out the example using ESP32-C6 and it worked fine.

dhrishi, 2024-01-11:
@kyberpunk Please close the issue if it is working fine at your end

kyberpunk, 2024-01-11:
Hi @dhrishi , I will test it to the end of the week again with building latest sources from scratch. It was strange that exactly the same use case was working with nRF52840.

kyberpunk, 2024-01-14:
Still don't know what exactly was wrong in the initial problem but with latest version of Home Assistant and Matter example it works. So I'm closing the issue, thanks for support.

closed 2024-01-14 by kyberpunk