project-chip/connectedhomeip #30108: [BUG] UNSUCCESSFUL BINDING Thermostat

BurievSardor, 2023-10-30:
### Reproduction steps

Ubuntu 22.04.1 LTS (jammy)
Linux 5.15.90.1-microsoft-standard-WSL2

I want to bind the  thermostat to the temperature sensor. 
But when i run the following command:
# ./out/debug/chip-tool binding write binding '[{"fabricIndex": 1, "node": 120, "endpoint": 1, "cluster": 402}]' 110 1

I got "0xc3 (UNSUPPORTED_CLUSTER)"  error.

110 - thermostat_node_ID (thermostat-app)

Any ideas?

### Bug prevalence

Whenever i do this

### GitHub hash of the SDK that was being used

de79f8bfc8897be4a344972d02f687b7d4bc20a3

### Platform

other

### Anything else?

[1698674747.883677][8642:8642] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_tool_kvs
[1698674747.884318][8642:8642] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1698674747.884355][8642:8642] CHIP:DL: ChipLinuxStorage::Init: Attempt to re-initialize with KVS config file: /tmp/chip_kvs
[1698674747.886807][8642:8642] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1698674747.886922][8642:8642] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1698674747.886953][8642:8642] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1698674747.887233][8642:8642] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-ykgCmI)
[1698674747.887531][8642:8642] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1698674747.887600][8642:8642] CHIP:DL: NVS set: chip-counters/reboot-count = 3 (0x3)
[1698674747.887757][8642:8642] CHIP:DL: Got Ethernet interface: eth0
[1698674747.887861][8642:8642] CHIP:DL: Found the primary Ethernet interface:eth0
[1698674747.888095][8642:8642] CHIP:DL: Failed to get WiFi interface
[1698674747.888134][8642:8642] CHIP:DL: Failed to reset WiFi statistic counts
[1698674747.888160][8642:8642] CHIP:IN: UDP::Init bind&listen port=0
[1698674747.888262][8642:8642] CHIP:IN: UDP::Init bound to port=53274
[1698674747.888312][8642:8642] CHIP:IN: UDP::Init bind&listen port=0
[1698674747.888347][8642:8642] CHIP:IN: UDP::Init bound to port=56574
[1698674747.888394][8642:8642] CHIP:IN: BLEBase::Init - setting/overriding transport
[1698674747.888436][8642:8642] CHIP:IN: TransportMgr initialized
[1698674747.888468][8642:8642] CHIP:FP: Initializing FabricTable from persistent storage
[1698674747.888650][8642:8642] CHIP:TS: Last Known Good Time: 2023-10-14T01:16:48
[1698674747.889861][8642:8642] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x30D3170C832CC98E, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1698674747.891151][8642:8642] CHIP:ZCL: Using ZAP configuration...
[1698674747.891594][8642:8642] CHIP:CTL: System State Initialized...
[1698674747.891678][8642:8642] CHIP:CTL: Setting attestation nonce to random value
[1698674747.891730][8642:8642] CHIP:CTL: Setting CSR nonce to random value
[1698674747.891828][8642:8642] CHIP:IN: UDP::Init bind&listen port=5550
[1698674747.892150][8642:8642] CHIP:IN: UDP::Init bound to port=5550
[1698674747.892200][8642:8642] CHIP:IN: UDP::Init bind&listen port=5550
[1698674747.892276][8642:8642] CHIP:IN: UDP::Init bound to port=5550
[1698674747.892402][8642:8642] CHIP:IN: TransportMgr initialized
[1698674747.893241][8642:8644] CHIP:DL: CHIP task running
[1698674747.893314][8642:8644] CHIP:DL: HandlePlatformSpecificBLEEvent 32784
[1698674747.893522][8642:8644] CHIP:CTL: Setting attestation nonce to random value
[1698674747.893624][8642:8644] CHIP:CTL: Setting CSR nonce to random value
[1698674747.894249][8642:8644] CHIP:CTL: Generating NOC
[1698674747.894709][8642:8644] CHIP:FP: Validating NOC chain
[1698674747.895521][8642:8644] CHIP:FP: NOC chain validation successful
[1698674747.895628][8642:8644] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1698674747.895668][8642:8644] CHIP:TS: Last Known Good Time: 2023-10-14T01:16:48
[1698674747.895675][8642:8644] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[... 101 lines cut ...]
[1698674748.134978][8642:8642] CHIP:CTL: Shutting down the commissioner
[1698674748.135065][8642:8642] CHIP:CTL: Shutting down the controller
[1698674748.135737][8642:8642] CHIP:IN: Expiring all sessions for fabric 0x1!!
[1698674748.135748][8642:8642] CHIP:IN: SecureSession[0x7f77f400afd0]: MarkForEviction Type:2 LSID:30250
[1698674748.135755][8642:8642] CHIP:SC: SecureSession[0x7f77f400afd0, LSID:30250]: State change 'kActive' --> 'kPendingEviction'
[1698674748.135762][8642:8642] CHIP:IN: SecureSession[0x7f77f400afd0]: Released - Type:2 LSID:30250
[1698674748.135772][8642:8642] CHIP:FP: Forgetting fabric 0x1
[1698674748.135810][8642:8642] CHIP:TS: Pending Last Known Good Time: 2023-10-14T01:16:48
[1698674748.135914][8642:8642] CHIP:TS: Previous Last Known Good Time: 2023-10-14T01:16:48
[1698674748.135922][8642:8642] CHIP:TS: Reverted Last Known Good Time to previous value
[1698674748.135943][8642:8642] CHIP:CTL: Shutting down the commissioner
[1698674748.135973][8642:8642] CHIP:CTL: Shutting down the controller
[1698674748.135980][8642:8642] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1698674748.136024][8642:8642] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1698674748.136057][8642:8642] CHIP:FP: Shutting down FabricTable
[1698674748.136066][8642:8642] CHIP:TS: Pending Last Known Good Time: 2023-10-14T01:16:48
[1698674748.136132][8642:8642] CHIP:TS: Previous Last Known Good Time: 2023-10-14T01:16:48
[1698674748.136138][8642:8642] CHIP:TS: Reverted Last Known Good Time to previous value
[1698674748.136258][8642:8642] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-3ZSwUN)
[1698674748.136471][8642:8642] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1698674748.136489][8642:8642] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1698674748.136497][8642:8642] CHIP:DL: Inet Layer shutdown
[1698674748.136503][8642:8642] CHIP:DL: BLE shutdown
[1698674748.136510][8642:8642] CHIP:DL: System Layer shutdown
[1698674748.139393][8642:8642] CHIP:TOO: Run command failure: IM Error 0x000005C3: General error: 0xc3 (UNSUPPORTED_CLUSTER)

bzbarsky-apple, 2023-10-31:
The example thermostat app does not have the Binding cluster enabled on endpoint 1, because it does not enable client clusters on that endpoint.

If you are adding a client, did you also add a Bindings server @BurievSardor ?

BurievSardor, 2023-10-31:
@bzbarsky-apple Hi! As I understand the thermostat needs to have a Binding cluster server that offers the binding service. But how to add a  Binding cluster server?

BurievSardor, 2023-10-31:
@bzbarsky-apple I think I found how to add a Binding server. When I run zap (to start the UI tool ) and open thermostat.zap I see for Endpoint-1 of Thermostat Binding server is not enabled.
[image]

BurievSardor, 2023-10-31:
I added a new Endpoint - 2  to examples/thermostat/thermostat-common/thermostat.zap by changing it

Endpoint -  1 :
Binding server -> enabled
Temperature cluster - client ->  enabled
Thermostat - server  -> disabled

Endpoint - 2:
Thermostat - server  -> enabled

After that removed examples/thermostat/thermostat-common/thermostat.matters then run following:
$ ./scripts/tools/zap/generate.py examples/thermostat/thermostat-common/thermostat.zap

110 - thermostat_node_ID (thermostat-app)

After that  compiled the thermostat app and commissioned the thermostat:
$ ./out/debug/chip-tool pairing onnetwork 110 20202021

then 

Run the following command:
$ ./out/debug/chip-tool binding write binding '[{"fabricIndex": 1, "node": 120, "endpoint": 1, "cluster": 402}]' 110 1

But I got this error:
[1698778684.574249][21999:21999] CHIP:TOO: Run command failure: IM Error 0x00000587: General error: 0x87 (CONSTRAINT_ERROR)

I don't why I got this error. Maybe it is because of that  I tried to bind the thermostat without commissioning the temperature sensor  into fabric

BurievSardor, 2023-11-01:
I removed  Endpoint - 2 and discarded changes of "examples/thermostat/thermostat-common/thermostat.zap".
I fixed 0x87 (CONSTRAINT_ERROR) by changing 402 to 1026. Because 402 was hex format. 

Then I added a Binding server and a TemperatureMeasurement client to Endpoint - 1. 

But  after  running the following command:
$ ./out/debug/chip-tool binding write binding '[{"fabricIndex": 1, "node": 120, "endpoint": 1, "cluster": 1026}]' 110 1
*******************
[1698851033.693221][5112:5112] CHIP:EM: Handling via exchange: 52610r, Delegate: 0x55fb682bb948
[1698851033.693247][5112:5112] CHIP:IM: Received Write request
[1698851033.693259][5112:5112] CHIP:DMG: IM WH moving to [Initialized]
[1698851033.693312][5112:5112] CHIP:DMG: WriteRequestMessage =
[1698851033.693373][5112:5112] CHIP:DMG: {
[1698851033.693435][5112:5112] CHIP:DMG:        suppressResponse = false,
[1698851033.693455][5112:5112] CHIP:DMG:        timedRequest = false,
[1698851033.693466][5112:5112] CHIP:DMG:        AttributeDataIBs =
[1698851033.693523][5112:5112] CHIP:DMG:        [
[1698851033.693536][5112:5112] CHIP:DMG:                AttributeDataIB =
[1698851033.693545][5112:5112] CHIP:DMG:                {
[1698851033.693566][5112:5112] CHIP:DMG:                        AttributePathIB =
[1698851033.693589][5112:5112] CHIP:DMG:                        {
[1698851033.693601][5112:5112] CHIP:DMG:                                Endpoint = 0x1,
[1698851033.693644][5112:5112] CHIP:DMG:                                Cluster = 0x1e,
[1698851033.693663][5112:5112] CHIP:DMG:                                Attribute = 0x0000_0000,
[1698851033.693677][5112:5112] CHIP:DMG:                        }
[1698851033.693688][5112:5112] CHIP:DMG:
[1698851033.693695][5112:5112] CHIP:DMG:                        Data = [
[1698851033.693702][5112:5112] CHIP:DMG:
[1698851033.693709][5112:5112] CHIP:DMG:                        ],
[1698851033.693753][5112:5112] CHIP:DMG:                },
[1698851033.693815][5112:5112] CHIP:DMG:
[1698851033.693831][5112:5112] CHIP:DMG:                AttributeDataIB =
[1698851033.693843][5112:5112] CHIP:DMG:                {
[1698851033.693855][5112:5112] CHIP:DMG:                        AttributePathIB =
[1698851033.693896][5112:5112] CHIP:DMG:                        {
[1698851033.693949][5112:5112] CHIP:DMG:                                Endpoint = 0x1,
[1698851033.693997][5112:5112] CHIP:DMG:                                Cluster = 0x1e,
[1698851033.694008][5112:5112] CHIP:DMG:                                Attribute = 0x0000_0000,
[1698851033.694016][5112:5112] CHIP:DMG:                                ListIndex = Null,
[1698851033.694025][5112:5112] CHIP:DMG:                        }
[1698851033.694034][5112:5112] CHIP:DMG:
[1698851033.694040][5112:5112] CHIP:DMG:                        Data =
[1698851033.694047][5112:5112] CHIP:DMG:                        {
[1698851033.694064][5112:5112] CHIP:DMG:                                0x1 = 120,
[1698851033.694111][5112:5112] CHIP:DMG:                                0x3 = 1,
[1698851033.694153][5112:5112] CHIP:DMG:                                0x4 = 1026,
[1698851033.694170][5112:5112] CHIP:DMG:                        },
[1698851033.694178][5112:5112] CHIP:DMG:                },
[1698851033.694217][5112:5112] CHIP:DMG:
[1698851033.694231][5112:5112] CHIP:DMG:        ],
[1698851033.694282][5112:5112] CHIP:DMG:
[1698851033.694291][5112:5112] CHIP:DMG:        moreChunkedMessages = false,
[1698851033.694343][5112:5112] CHIP:DMG:        InteractionModelRevision = 11
[1698851033.694354][5112:5112] CHIP:DMG: },
[1698851033.694455][5112:5112] CHIP:DMG: AccessControl: checking f=1 a=c s=0x000000000001B669 t= c=0x0000_001E e=1 p=m
[1698851033.694521][5112:5112] CHIP:DMG: AccessControl: allowed
[1698851033.694554][5112:5112] CHIP:DMG: Endpoint 1, Cluster 0x0000_001E update version to b1332db3
[1698851033.694618][5112:5112] CHIP:DMG: IM WH moving to [AddStatus]
Segmentation fault
************************

Thermostat-app crashes with Segmentation fault. I did not add BindingHandler class. Is it  because of that?

BurievSardor, 2023-11-01:
I will close this issue and open a new issue on the segmentation fault