project-chip/connectedhomeip #30116: [BUG] The bridge-app can connect to the homepod but not the google nest hub

nmtoan91, 2023-10-31:
### Reproduction steps

The bridge can easily pair with my HomePod, but there's no way it can connect to my Google Nest Hub 2.
The error is too general "There is an error when connecting to the device".

### Bug prevalence

all the time

### GitHub hash of the SDK that was being used

latest

### Platform

darwin

lecndav, 2023-10-31:
Have you set up a matter integration project in the console?

nmtoan91, 2023-11-07:
I used the QR code that was generated from matter.

lecndav, 2023-11-07:
You need to set up a matter integration project in the google cloud console before, in order that the google home app gets the required information.

lecndav, 2023-11-08:
I stumbled over a tutorial from silabs https://docs.silabs.com/matter/2.1.0/matter-bridge/readme-user#using-google-nest-hub