project-chip/connectedhomeip #30142: [BUG] conformance issues on air quality app

cecille, 2023-11-01:
### Reproduction steps

Conformance test

[2023-10-31 22:13:54.711469][TEST][STDOUT][MatterTest] 10-31 22:13:54.707 INFO 
[2023-10-31 22:13:54.711482][TEST][STDOUT]Problem: ProblemSeverity.ERROR
[2023-10-31 22:13:54.711492][TEST][STDOUT]    test_name: test_spec_conformance
[2023-10-31 22:13:54.711502][TEST][STDOUT]    location: 
[2023-10-31 22:13:54.711511][TEST][STDOUT]        Endpoint: 0,
[2023-10-31 22:13:54.711523][TEST][STDOUT]        Cluster:  40 (0x28) BasicInformation,
[2023-10-31 22:13:54.711533][TEST][STDOUT]        Attribute:21 (0x15)
[2023-10-31 22:13:54.711543][TEST][STDOUT]    problem: Attribute 0x15 is required, but is not present on the DUT. Conformance: M, implemented features: 
[2023-10-31 22:13:54.711554][TEST][STDOUT]    spec_location: 
[2023-10-31 22:13:54.711564][TEST][STDOUT]
[2023-10-31 22:13:54.711574][TEST][STDOUT][MatterTest] 10-31 22:13:54.707 INFO 
[2023-10-31 22:13:54.711586][TEST][STDOUT]Problem: ProblemSeverity.ERROR
[2023-10-31 22:13:54.711596][TEST][STDOUT]    test_name: test_spec_conformance
[2023-10-31 22:13:54.711607][TEST][STDOUT]    location: 
[2023-10-31 22:13:54.711617][TEST][STDOUT]        Endpoint: 0,
[2023-10-31 22:13:54.711627][TEST][STDOUT]        Cluster:  40 (0x28) BasicInformation,
[2023-10-31 22:13:54.711637][TEST][STDOUT]        Attribute:22 (0x16)
[2023-10-31 22:13:54.711647][TEST][STDOUT]    problem: Attribute 0x16 is required, but is not present on the DUT. Conformance: M, implemented features: 
[2023-10-31 22:13:54.711658][TEST][STDOUT]    spec_location: 
[2023-10-31 22:13:54.711668][TEST][STDOUT]
[2023-10-31 22:13:54.711678][TEST][STDOUT][MatterTest] 10-31 22:13:54.707 INFO 
[2023-10-31 22:13:54.711690][TEST][STDOUT]Problem: ProblemSeverity.WARNING
[2023-10-31 22:13:54.711700][TEST][STDOUT]    test_name: test_spec_conformance
[2023-10-31 22:13:54.711710][TEST][STDOUT]    location: 
[2023-10-31 22:13:54.711720][TEST][STDOUT]       Endpoint: 0,
[2023-10-31 22:13:54.711730][TEST][STDOUT]       Cluster:  42 (0x2a) OtaSoftwareUpdateRequestor
[2023-10-31 22:13:54.711740][TEST][STDOUT]    problem: Standard cluster found on device, but is not present in spec data
[2023-10-31 22:13:54.711751][TEST][STDOUT]    spec_location: 
[2023-10-31 22:13:54.711760][TEST][STDOUT]
[2023-10-31 22:13:54.711770][TEST][STDOUT][MatterTest] 10-31 22:13:54.707 INFO 
[2023-10-31 22:13:54.711782][TEST][STDOUT]Problem: ProblemSeverity.ERROR
[2023-10-31 22:13:54.711793][TEST][STDOUT]    test_name: test_spec_conformance
[2023-10-31 22:13:54.711803][TEST][STDOUT]    location: 
[2023-10-31 22:13:54.711813][TEST][STDOUT]        Endpoint: 0,
[2023-10-31 22:13:54.711825][TEST][STDOUT]        Cluster:  48 (0x30) GeneralCommissioning,
[2023-10-31 22:13:54.711835][TEST][STDOUT]        Attribute:65532 (0xfffc)
[2023-10-31 22:13:54.711845][TEST][STDOUT]    problem: Unknown feature with mask 0x02
[2023-10-31 22:13:54.711855][TEST][STDOUT]    spec_location: 
[2023-10-31 22:13:54.711865][TEST][STDOUT]
[2023-10-31 22:13:54.711875][TEST][STDOUT][MatterTest] 10-31 22:13:54.707 INFO 
[2023-10-31 22:13:54.711887][TEST][STDOUT]Problem: ProblemSeverity.ERROR
[2023-10-31 22:13:54.711897][TEST][STDOUT]    test_name: test_spec_conformance
[2023-10-31 22:13:54.711907][TEST][STDOUT]    location: 
[2023-10-31 22:13:54.711917][TEST][STDOUT]        Endpoint: 0,
[2023-10-31 22:13:54.711927][TEST][STDOUT]        Cluster:  48 (0x30) GeneralCommissioning,
[2023-10-31 22:13:54.711936][TEST][STDOUT]        Attribute:65532 (0xfffc)
[2023-10-31 22:13:54.711947][TEST][STDOUT]    problem: Unknown feature with mask 0x04
[2023-10-31 22:13:54.711957][TEST][STDOUT]    spec_location: 
[2023-10-31 22:13:54.711967][TEST][STDOUT]
[2023-10-31 22:13:54.711977][TEST][STDOUT][MatterTest] 10-31 22:13:54.707 INFO 
[2023-10-31 22:13:54.711989][TEST][STDOUT]Problem: ProblemSeverity.ERROR
[2023-10-31 22:13:54.712001][TEST][STDOUT]    test_name: test_spec_conformance
[2023-10-31 22:13:54.712010][TEST][STDOUT]    location: 
[2023-10-31 22:13:54.712020][TEST][STDOUT]        Endpoint: 0,
[2023-10-31 22:13:54.712030][TEST][STDOUT]        Cluster:  49 (0x31) NetworkCommissioning,
[2023-10-31 22:13:54.712040][TEST][STDOUT]        Attribute:2 (0x02)
[2023-10-31 22:13:54.712050][TEST][STDOUT]    problem: Attribute 0x02 is included, but is disallowed by conformance. Conformance: WI | TH, implemented features: ET
[2023-10-31 22:13:54.712062][TEST][STDOUT]    spec_location: 
[2023-10-31 22:13:54.712072][TEST][STDOUT]
[2023-10-31 22:13:54.712082][TEST][STDOUT][MatterTest] 10-31 22:13:54.707 INFO 
[2023-10-31 22:13:54.712094][TEST][STDOUT]Problem: ProblemSeverity.ERROR
[2023-10-31 22:13:54.712103][TEST][STDOUT]    test_name: test_spec_conformance
[2023-10-31 22:13:54.712113][TEST][STDOUT]    location: 
[2023-10-31 22:13:54.712124][TEST][STDOUT]        Endpoint: 0,
[2023-10-31 22:13:54.712133][TEST][STDOUT]        Cluster:  49 (0x31) NetworkCommissioning,
[2023-10-31 22:13:54.712143][TEST][STDOUT]        Attribute:3 (0x03)
[... 709 lines cut ...]
[2023-10-31 22:13:54.717875][TEST][STDOUT]    problem: Attribute 0x09 is required, but is not present on the DUT. Conformance: M, implemented features: MEA,LEV,MED,CRI,PEA,AVG
[2023-10-31 22:13:54.717882][TEST][STDOUT]    spec_location: 
[2023-10-31 22:13:54.717887][TEST][STDOUT]
[2023-10-31 22:13:54.717892][TEST][STDOUT][MatterTest] 10-31 22:13:54.710 INFO 
[2023-10-31 22:13:54.717899][TEST][STDOUT]Problem: ProblemSeverity.ERROR
[2023-10-31 22:13:54.717904][TEST][STDOUT]    test_name: test_spec_conformance
[2023-10-31 22:13:54.717910][TEST][STDOUT]    location: 
[2023-10-31 22:13:54.717915][TEST][STDOUT]        Endpoint: 1,
[2023-10-31 22:13:54.717921][TEST][STDOUT]        Cluster:  1071 (0x42f) RadonConcentrationMeasurement,
[2023-10-31 22:13:54.717927][TEST][STDOUT]        Attribute:10 (0x0a)
[2023-10-31 22:13:54.717934][TEST][STDOUT]    problem: Attribute 0x0a is required, but is not present on the DUT. Conformance: LEV, implemented features: MEA,LEV,MED,CRI,PEA,AVG
[2023-10-31 22:13:54.717939][TEST][STDOUT]    spec_location: 
[2023-10-31 22:13:54.717945][TEST][STDOUT]

### Bug prevalence

always

### GitHub hash of the SDK that was being used

fe513bbf0d3624be3f7c1f86bee55210e5418196

### Platform

other

cecille, 2023-11-01:
Ignore the error about RelativeHumidityMeasurement for now - this is just a problem with data missing from the scraper

mhazley, 2023-11-01:
@tcave-bb are you able to have a look at this?

tcave-bb, 2023-11-03:
Yes, I'll look at it on Monday.

tcave-bb, 2024-01-24:
@cecille , Could you check PR https://github.com/project-chip/connectedhomeip/pull/31625, this passes all checks. Can you confirm the conformance test is ran as part of the checks? Or should this be done with a separate tool?

cecille, 2024-01-25:
Sorry, commented on the PR before the issue. Conformance is not currently run for this app in the CI because it would not have passed. I don't even think you're compiling this app in the CI.