project-chip/connectedhomeip #30151: [BUG] FailSafe timer fails commissioning early, "Can't reach Google," network works anyway

emmby, 2023-11-01:
### Reproduction steps

I have an ESP32 using Matter 08b13661b6 (July 19 2023). Whenever I try to commission a test device onto Google Home, the FailSafeTimer expires well before 60 seconds (from section 5.5 of [the 1.0 spec](https://csa-iot.org/wp-content/uploads/2022/11/22-27349-001_Matter-1.0-Core-Specification.pdf)) have elapsed. See below where it expires in less than 45 seconds. When this happens, Google Home reports "Can't reach Google" and aborts the commissioning after the "Generating Matter Credentials," "Testing Network Connectivity", and "Connecting device to Google Home" steps.

[Troubleshooting for that error says](https://developers.home.google.com/matter/troubleshooting#commissioning_fails_with_a_could_not_contact_google_error_message) to check that I've create the proper PID/VID on the developer console, which I have and was working previously.

The funny thing is that even though commissioning has failed, the device can access the network just fine. `chip::DeviceLayer::ConfigurationMgr().IsFullyProvisioned()` reports true and network requests go through. But `FabricTable.FabricCount()` still reports 0.

The Google Home error message is new, I was not seeing that behavior previously even for the same device codebase. I'm wondering whether I got my fabric into a weird state somehow but I'm also not sure how to reset it. Re-installing the Google Home app didn't resolve the problem.

Why won't this device commission onto the fabric?

* ESP32 Pico on Adafruit Feather ESP32 v2 using Wifi
* Yacubane esp32-arduino-matter v1.0.0-beta.6
* Matter 08b13661b6
* esp-matter	`603296a`
* esp-idf	v4.4.5
* arduino-esp32	2.0.11

```
07:58:51.646 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
07:58:51.646 -> configsip: 271414342, SPIWP:0xee
07:58:51.646 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
07:58:51.683 -> mode:DIO, clock div:1
07:58:51.683 -> load:0x3fff0030,len:1184
07:58:51.683 -> load:0x40078000,len:13260
07:58:51.683 -> load:0x40080400,len:3028
07:58:51.683 -> entry 0x400805e4
07:58:53.656 -> E (1301) esp_matter_providers: custom_dac_provider cannot be NULL
07:58:54.457 -> E (3862) chip[DL]: Long dispatch time: 780 ms, for event type 2
07:58:54.488 -> E (3908) chip[ZCL]: Duplicate attribute override registration failed
07:59:15.070 -> E (24482) chip[DL]: Long dispatch time: 1514 ms, for event type 7
07:59:19.728 -> E (29125) chip[DL]: Long dispatch time: 979 ms, for event type 7
07:59:23.659 -> E (33047) chip[DL]: Long dispatch time: 1345 ms, for event type 7
07:59:24.027 -> E (33447) wifi:sta is connecting, return error
07:59:24.027 -> E (33448) chip[DL]: esp_wifi_connect() failed: ESP_ERR_WIFI_CONN
07:59:24.059 -> E (33455) wifi:sta is connecting, return error
07:59:24.059 -> E (33459) chip[DL]: esp_wifi_connect() failed: ESP_ERR_WIFI_CONN
07:59:24.059 -> E (33466) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
07:59:25.643 -> E (35037) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
07:59:25.643 -> E (35044) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
07:59:25.742 -> E (35140) chip[DL]: Failed to get configured network when updating network status: Error ESP32:0x0500300F
07:59:28.703 -> E (38097) chip[DL]: Long dispatch time: 923 ms, for event type 3
07:59:28.802 -> E (38222) chip[SC]: The device does not support GetClock_RealTimeMS() API: 6c.  Falling back to Last Known Good UTC Time
07:59:30.150 -> E (39576) chip[DL]: Long dispatch time: 1323 ms, for event type 2
07:59:33.011 -> E (42422) chip[DMG]: Error retrieving data from clusterId: 0x0000_0031, err = b
07:59:33.172 -> E (42580) chip[DMG]: Error retrieving data from clusterId: 0x0000_0028, err = b
07:59:33.334 -> E (42751) chip[DMG]: Error retrieving data from clusterId: 0x0000_001D, err = b
07:59:35.713 -> E (45133) chip[SVR]: Failsafe timer expired
07:59:35.713 -> E (45134) chip[SVR]: Commissioning failed (attempt 1): 32
07:59:35.745 -> E (45138) chip[BLE]: CancelBleIncompleteConnection() failed, err = 3
07:59:35.810 -> E (45205) chip[ZCL]: OpCreds: Got FailSafeTimerExpired
07:59:35.810 -> E (45207) chip[ZCL]: OpCreds: Proceeding to FailSafeCleanup on fail-safe expiry!
07:59:35.810 -> E (45222) chip[FP]: Reverting pending fabric data for fabric 0x1
07:59:35.810 -> E (45240) chip[FP]: Warning: metadata not found during delete of fabric 0x1
07:59:35.980 -> E (45397) chip[FP]: Warning: metadata not found during delete of fabric 0x1
07:59:35.980 -> E (45405) chip[ZCL]: OpCreds: failed to delete fabric at index 1: d8
07:59:35.980 -> E (45406) chip[DL]: ble_gap_adv_set_data failed: Error ESP32:0x05B0001E 3840
07:59:36.013 -> E (45411) chip[DL]: Configure Adv Data failed: Error ESP32:0x05B0001E
07:59:36.013 -> E (45417) chip[DL]: Disabling CHIPoBLE service due to error: Error ESP32:0x05B0001E
07:59:43.318 -> E (52746) esp_matter_core: Node already exists
07:59:43.351 -> E (52767) esp_matter_core: esp_matter has started
```

### Bug prevalence

Every time recently, never previously

### GitHub hash of the SDK that was being used

08b13661b6

### Platform

esp32

### Platform Version(s)

esp-matter 603296a

### Anything else?

[image]

bzbarsky-apple, 2023-11-01:
@emmby That's not much to go on in the way of logs...  In particular: creating a PASE session arms a fail-safe for 60 seconds, but the client can then re-arm it for a different value.  The log is not showing any incoming commands; are there none, or are they just not being logged?

emmby, 2023-11-01:
Sorry, better logs:

```
Setting temperature attribute to 2375 (23.8C)
State: entered main_region_TemperatureSynced
State: entered main_region_SyncToRTC
State: entered main_region_DisplayClock
State: entered main_region_Sleep
D (138356) chip[DIS]: Warning: Attempt to mDNS broadcast failed on st1:  Error LwIP:0x03000001
D (138358) chip[DIS]: mDNS broadcast had only partial success: 1 successes and 1 failures.
D (138445) chip[DIS]: Warning: Attempt to mDNS broadcast failed on st1:  Error LwIP:0x03000001
D (138447) chip[DIS]: mDNS broadcast had only partial success: 1 successes and 1 failures.
D (138455) chip[DIS]: Warning: Attempt to mDNS broadcast failed on st1:  Error LwIP:0x03000001
D (138460) chip[DIS]: Warning: Attempt to mDNS broadcast failed on st1:  Error LwIP:0x03000001
I (138467) chip[DIS]: mDNS broadcast full failed in 2 separate send attempts.
E (138586) chip[DL]: Failed to post event to CHIP Platform event queue
E (138587) chip[CSL]: Failed to queue CHIP System Layer lambda event: 1000000
: factory
I (142945) chip[DL]: WIFI_EVENT_STA_DISCONNECTED
I (142946) chip[DL]: Starting ESP WiFi layer
E (142946) chip[DL]: esp_wifi_start() failed: ESP_ERR_WIFI_STOP_STATE
I (142949) chip[DL]: WIFI_EVENT_STA_STOP
I (142953) chip[DL]: Starting ESP WiFi layer
E (142957) chip[DL]: esp_wifi_start() failed: ESP_ERR_WIFI_STOP_STATE
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
Wakeup was not caused by deep sleep: 0
Found PCT2075 temperature chip
State: entered main_region_HardwareInitialized
D (1251) esp_matter_core: Set val called with string with size 0
D (1252) esp_matter_core: Set val called with string with size 0
D (1255) esp_matter_core: Set val called with string with size 0
D (1261) esp_matter_core: Set val called with string with size 0
D (1267) esp_matter_core: Set val called with string with size 0
D (1273) esp_matter_core: Set val called with string with size 0
D (1279) esp_matter_core: Store attribute in nvs: endpoint_id-0x0, cluster_id-0x28, attribute_id-0x6
D (1287) esp_matter_core: Set val called with string with size 0
D (1293) esp_matter_core: Set val called with string with size 0
D (1299) esp_matter_core: Set val called with string with size 0
D (1304) esp_matter_core: Set val called with string with size 0
D (1310) esp_matter_core: Set val called with string with size 0
D (1316) esp_matter_core: read attribute from nvs: endpoint_id-0x0, cluster_id-0x28, attribute_id-0x5
D (1325) esp_matter_core: Store attribute in nvs: endpoint_id-0x0, cluster_id-0x28, attribute_id-0x5
D (1335) esp_matter_core: Set val called with string with size 0
D (1340) esp_matter_core: Set val called with string with size 0
D (1345) esp_matter_core: read attribute from nvs: endpoint_id-0x0, cluster_id-0x31, attribute_id-0x4
D (1354) esp_matter_core: Store attribute in nvs: endpoint_id-0x0, cluster_id-0x31, attribute_id-0x4
D (1364) esp_matter_core: Set val called with string with size 0
D (1369) esp_matter_core: Set val called with string with size 0
D (1374) esp_matter_core: read attribute from nvs: endpoint_id-0x0, cluster_id-0x33, attribute_id-0x1
D (1383) esp_matter_core: Store attribute in nvs: endpoint_id-0x0, cluster_id-0x33, attribute_id-0x1
D (1394) esp_matter_core: read attribute from nvs: endpoint_id-0x0, cluster_id-0x3e, attribute_id-0x0
D (1401) esp_matter_core: Set val called with string with size 0
D (1407) esp_matter_core: Store attribute in nvs: endpoint_id-0x0, cluster_id-0x3e, attribute_id-0x0
D (1416) esp_matter_core: read attribute from nvs: endpoint_id-0x0, cluster_id-0x3e, attribute_id-0x3
D (1425) esp_matter_core: Store attribute in nvs: endpoint_id-0x0, cluster_id-0x3e, attribute_id-0x3
D (1435) esp_matter_core: read attribute from nvs: endpoint_id-0x0, cluster_id-0x3e, attribute_id-0x1
D (1442) esp_matter_core: Set val called with string with size 0
D (1448) esp_matter_core: Store attribute in nvs: endpoint_id-0x0, cluster_id-0x3e, attribute_id-0x1
D (1457) esp_matter_core: read attribute from nvs: endpoint_id-0x0, cluster_id-0x3e, attribute_id-0x4
D (1466) esp_matter_core: Set val called with string with size 0
D (1472) esp_matter_core: Store attribute in nvs: endpoint_id-0x0, cluster_id-0x3e, attribute_id-0x4
[... 1944 lines cut ...]
D (46667) chip[DMG]: Endpoint 0, Cluster 0x0000_003E update version to 5b3c719e
D (46669) chip[DMG]: Endpoint 0, Cluster 0x0000_003E update version to 5b3c719f
D (46681) chip[DL]: Using hash:accaec5f6ac0c34 for nvs key:f/1/s/000000005356B4C2
D (46691) chip[DL]: Using hash:00f9871538fa9e7 for nvs key:g/s/BQpIEH30QHsT48nD/AY49A==
D (46778) chip[DL]: Using hash:accaec5f6ac0c34 for nvs key:f/1/s/000000005356B4C2
I (47298) chip[TS]: Pending Last Known Good Time: 2023-11-01T23:18:28
I (47299) chip[TS]: Previous Last Known Good Time: 2023-07-24T23:38:47
I (47300) chip[TS]: Reverted Last Known Good Time to previous value
D (47307) chip[EVL]: LogEvent event number: 0x0000000000000005 priority: 1, endpoint id:  0x0 cluster id: 0x0000_0028 event id: 0x2 Sys timestamp: 0x000000000000A6DA
D (47321) chip[DMG]: All ReadHandler-s are clean, clear GlobalDirtySet
E (47407) chip[FP]: Warning: metadata not found during delete of fabric 0x1
E (47487) chip[ZCL]: OpCreds: failed to delete fabric at index 1: d8
[... cut ...]
Commissioning failed, fail safe timer expired
E (47511) chip[DL]: Long dispatch time: 1688 ms, for event type 32783
I (47512) chip[FS]: Fail-safe cleanly disarmed
E (47515) chip[DL]: ble_gap_adv_set_data failed: Error ESP32:0x05B0001E 3840
E (47521) chip[DL]: Configure Adv Data failed: Error ESP32:0x05B0001E
E (47528) chip[DL]: Disabling CHIPoBLE service due to error: Error ESP32:0x05B0001E
Commissioning window opened
Fabric will be removed
Fabric removed successfully
Fabric will be removed
```

bzbarsky-apple, 2023-11-02:
OK, so that log makes it clear what happened:
```
I (44816) chip[FS]: GeneralCommissioning: Received ArmFailSafe (1s)
```

Why did the client decide to arm a 1-second failsafe?  Sounds like an issue you should raise with them, not a core Matter issue.

closed 2023-11-02 by bzbarsky-apple

KEEPHEADU, 2024-08-16:
Hi @emmby , I met the same problem, may I ask how you finally solved it? 
```
[00:00:17.348][info  ][FS] GeneralCommissioning: Received ArmFailSafe (1s)

[00:00:17.348][detail][DMG] Endpoint 0, Cluster 0x0000_0030 update version to 57ce0b7d

[00:00:17.349[00:00:17.349][detail][DMG] Command handler moving to [AddedComma]

[00:00:17.349][detail][DMG] Decreasing reference count for CommandHandler, remaining 0

[00:00:17.350][info  ][EM] <<< [E:48972r S:52584 M:211344138] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [BLE] --- Type 0001:09 (IM:InvokeCommand[00:00:17.442][info  ][DL] Tx Confirmation received

[00:00:17.442][info  ][DL]  stop soft timer

[00:00:17.443][info  ][DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm

[00:00:18.348][info  ][FS] Fail-safe timer expired

[00:00:18.348][info  ][DL] _OnPlatformEvent default:  event->Type = 32784

[00:00:18.348][error ][SVR] Failsafe timer expired

[00:00:18.348][detail][IN] SecureSession[0x20004e30]: MarkForEviction Type:1 LSID:52584

[00:00:18.348][info  ][SC] SecureSession[0x20004e30, LSID:52584]: State change 'kActive' --> 'kPendingEviction'

[00:00:18.348][detail][IN] SecureSession[0x20004e30]: Released - Type:1 LSID:52584

[00:00:18.348][error ][SVR] Commissioning failed (attempt 1): 32
```