project-chip/connectedhomeip #30164: [BUG] Non-backwards-compatible breaking change in Application Launcher cluster.

bzbarsky-apple, 2023-11-02:
### Reproduction steps

https://github.com/project-chip/connectedhomeip/pull/30134 made a breaking change to client APIs by changing a writable attribute to readonly.  

We need to either revert that change or figure out some other way of making the APIs continue to exist.

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

6ec33d400734ac55ee4b200fece52896398f7f30

### Platform

darwin

bzbarsky-apple, 2023-11-02:
@andy31415

andy31415, 2023-11-02:
What is the breaking change (i.e. what methods are missing as a result)? I had assumed this was not an ABI issue.

If this is just some `write*` method availability, could we on-off them into some h/mm file?

andy31415, 2023-11-02:
I can see no reasonable way to change the spec to make this attribute writable (what would be the meaning? Syaing `SHALL always return an error` also seems wrong) nor do I think keeping it writable in the sdk not in the spec is that practical.

bzbarsky-apple, 2023-11-02:
> If this is just some write* method availability, could we on-off them into some h/mm file?

That is what it is, and maybe we can, but this is breaking things in the meantime.  I am happy to look into doing this change in a non-breaking way, but I can't do it this week.

PR #30174 (2023-11-02): Add a "Darwin force write" configuration for darwin APIs
Darwin API rules are `you shall not remove anything` and exceptions are hard to obtain.

Until we figure out what to do, add an exception support.

Fixes #30164

## Tested

changed attribute to writable=true and noticed there were no diffs in `src/darwin/Framework/CHIP/`
closed 2023-11-02 by andy31415