project-chip/connectedhomeip #30168: [BUG] Fail to pair to to partner app because it doesn't have a staged identifier

robinmo, 2023-11-02:
### Reproduction steps

Version 15.0 (15A240d)
iPhone 8/ 16.7.1 （same as 16.3）
enter the manual code
1.user MatterSupport to commission first and then commission third-party fabric
2.after set name pop up(MatterSupport UI), failed to pair.
the core error log said `it doesn't have a staged identifier`
```
Successfully finished commissioning step 'Cleanup'
Nov  2 17:02:57 iPhone8-fujianyonggeren HomeUIService[23499] <Notice>: -[HSSetupStateMachineCHIPPartnerConfiguration pairStagedCHIPAccessoryToPartnerAppHome] pairingIdentifier <private>, setupCode <private>, ecosystem <private>. Unless this is an accessory previously paired to system commissioner, it will fail to pair to to partner app because it doesn't have a staged identifier
Nov  2 17:03:12 iPhone8-fujianyonggeren runningboardd(RunningBoard)[33] <Notice>: Acquiring assertion targeting [xpcservice<com.tplink.aria.commissionRequestHandler([application<com.apple.Home.HomeUIService>:23499])>:23501] from originator [application<com.apple.Home.HomeUIService>:23499] with description <RBSAssertionDescriptor| "com.apple.extension.session" ID:33-23499-72897 target:23501 attributes:[	<RBSLegacyAttribute| requestedReason:ViewService reason:ViewService flags:( AllowIdleSleep PreventTaskSuspend PreventTaskThrottleDown WantsForegroundResourcePriority )>,
Nov  2 17:03:12 iPhone8-fujianyonggeren runningboardd(RunningBoard)[33] <Notice>: Assertion 33-23499-72897 (target:[xpcservice<com.tplink.aria.commissionRequestHandler([application<com.apple.Home.HomeUIService>:23499])>:23501]) will be created as active
Nov  2 17:03:12 iPhone8-fujianyonggeren runningboardd(RunningBoard)[33] <Notice>: Calculated state for xpcservice<com.tplink.aria.commissionRequestHandler([application<com.apple.Home.HomeUIService>:23499])>: running-active (role: UserInteractiveNonFocal)
Nov  2 17:03:12 iPhone8-fujianyonggeren runningboardd(RunningBoard)[33] <Notice>: [xpcservice<com.tplink.aria.commissionRequestHandler([application<com.apple.Home.HomeUIService>:23499])>:23501] Set jetsam priority to 100 [0] flag[1]
Nov  2 17:03:12 iPhone8-fujianyonggeren runningboardd(RunningBoard)[33] <Notice>: [xpcservice<com.tplink.aria.commissionRequestHandler([application<com.apple.Home.HomeUIService>:23499])>:23501] Resuming task.
Nov  2 17:03:12 iPhone8-fujianyonggeren runningboardd(RunningBoard)[33] <Notice>: [xpcservice<com.tplink.aria.commissionRequestHandler([application<com.apple.Home.HomeUIService>:23499])>:23501] Result 45 setting darwin role to UserInteractiveNonFocal: Operation not supported, falling back to setting priority
Nov  2 17:03:12 iPhone8-fujianyonggeren runningboardd(RunningBoard)[33] <Notice>: [xpcservice<com.tplink.aria.commissionRequestHandler([application<com.apple.Home.HomeUIService>:23499])>:23501] Set darwin priority to: PRIO_DEFAULT
Nov  2 17:03:12 iPhone8-fujianyonggeren runningboardd(RunningBoard)[33] <Notice>: [xpcservice<com.tplink.aria.commissionRequestH
```
[lack_of_a_staged_identifier.log](https://github.com/project-chip/connectedhomeip/files/13237245/lack_of_a_staged_identifier.log)

Anyway for my iPad2Pro 17.0.3 manual code commission is OK

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

9c0ca13 - Fix Linux standalone job. (#23897)

### Platform

darwin

### Platform Version(s)

iOS 16.7

robinmo, 2023-11-02:
@bzbarsky-apple could you have any debug suggestion

bzbarsky-apple, 2023-11-02:
@robinmo Please file a Feedback Assistant ticket per the instructions below.  That will have better logs (with the logging profiles it says to install).  Please put the ticket ID here once it's filed and I can take a look.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

robinmo, 2023-11-08:
I try to make a feedback today, but found it success for manual code pairing🤣
same build same iPhone with same version. confused.

bzbarsky-apple, 2023-11-14:
@robinmo I don't know what to tell you.  Presumably something was different between the success and fail cases, but I don't have any data to tell you what.

robinmo, 2023-11-21:
> @robinmo I don't know what to tell you. Presumably something was different between the success and fail cases, but I don't have any data to tell you what.

I found on 17.1.1 and 17.2 beta3, it will failed to create noc, the same code works on 17.0

robinmo, 2023-12-02:
> @robinmo Which exact API will "failed to create noc"? Given what inputs? What is the error that is returned?

Sorry about that, I found that it is vendorID 0 causing the failure.After blocking vendorID 0, it success.

bzbarsky-apple, 2023-12-06:
Blocking it from what?  VendorID == 0 can happen for various reasons in different situations.  Where exactly are you seeing it, and why is it being a problem there @robinmo ?

robinmo, 2023-12-06:
@bzbarsky-apple 
A short manual code.
In MTRDeviceController.mm
```
- (BOOL)startup:(MTRDeviceControllerStartupParamsInternal *)startupParams
{
    __block BOOL commissionerInitialized = NO;
    if ([self isRunning]) {
        MTR_LOG_ERROR("Unexpected duplicate call to startup");
        return NO;
    }

    dispatch_sync(_chipWorkQueue, ^{
        if ([self isRunning]) {
            return;
        }
// here block 0 vendorID
        if (startupParams.vendorID == nil || [startupParams.vendorID unsignedShortValue] == chip::VendorId::Common) {
            // Shouldn't be using the "standard" vendor ID for actual devices.
            MTR_LOG_ERROR("%@ is not a valid vendorID to initialize a device controller with", startupParams.vendorID);
            return;
        }

        if (startupParams.operationalCertificate == nil && startupParams.nodeID == nil) {
            MTR_LOG_ERROR("Can't start a controller if we don't know what node id it is");
            return;
        }

        if ([startupParams keypairsMatchCertificates] == NO) {
            MTR_LOG_ERROR("Provided keypairs do not match certificates");
            return;
        }
```

but startupParams API has no comment on this case(0 will be blocked)
```
/**
 * Vendor ID (allocated by the Connectivity Standards Alliance) for
 * this controller.
 *
 * If not nil, must not be the "standard" vendor id (0).
 *
 * When creating a new fabric:
 *
 * * Must not be nil.
 *
 * When using an existing fabric:
 *
 * * Will override existing value if not nil. Otherwise existing value will be
 *   used.
 */
@property (nonatomic, copy, nullable) NSNumber * vendorID MTR_AVAILABLE(ios(16.4), macos(13.3), watchos(9.4), tvos(16.4));
```

bzbarsky-apple, 2023-12-06:
That comment clearly says:

>  * If not nil, must not be the "standard" vendor id (0).

bzbarsky-apple, 2023-12-06:
@robinmo So is the only issue here that you were trying to start up a controller using vendor id 0?

robinmo, 2023-12-09:
> @robinmo So is the only issue here that you were trying to start up a controller using vendor id 0?

Yes, by avoiding setting 0 vendorID, the whole commission is ok.

closed 2024-01-31 by robinmo

huchuankan, 2024-03-14:
Matter Commission  demo video ： https://www.bilibili.com/video/BV1Nx421y7sa/