project-chip/connectedhomeip #30170: [ACL-2.7] LastReceivedEventNumber seems to be not working

manjunath-grl, 2023-11-02:
### Reproduction steps

Summary: 
When the ACL-2.5 and 2.7 tests are run consecutively, ACL-2.7 fails at step 9 "AccessControlExtensionChanged" event read step response contains the expected value but got "Test Failure : The test expects 1 responses but got 7 responses".

Reproduction step:
Terminal 1: 
Launch `./chip-all-clusters-app `

Terminal 2:
Step 1: Pairing  `./out/chip-tool/chip-tool pairing code 0x12344321 MT:-24J0AFN00KA0648G00`
Step 2: Run tests ACL-2.5 & 2.7
`./scripts/tests/yaml/chiptool.py tests Test_TC_ACL_2_5`
`./scripts/tests/yaml/chiptool.py tests Test_TC_ACL_2_7`

Diff link: https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:ACL_2_7_AccessControlExtensionChanged_Event_check_failure

### Bug prevalence

Everytime when executed

### GitHub hash of the SDK that was being used

bfc8c327826adfbdfe520fe207f5e4d203462887

### Platform

raspi, other

### Anything else?

Screenshot: 
[image]
[image]
Execution logs: 
[ACL_25_27_logs.txt](https://github.com/project-chip/connectedhomeip/files/13237578/ACL_25_27_logs.txt)

bzbarsky-apple, 2023-11-02:
LastReceivedEventNumber is the last number the test saw, right?  If the test has not seen any events, it presumably sends 0... but if the DUT is not rebooted between tests, it will have the events from the previous test.

raul-marquez-csa, 2025-06-29:
Python ports for ACL TCs 2.3, 2.4, 2.5, 2.6, 2.7, and 2.8 have been created and merged.

[[Test] Creating ACL_2_3 python3 test module #38493](https://github.com/project-chip/connectedhomeip/pull/38493)
[Creating ACL 2 4 python3 test module #38314](https://github.com/project-chip/connectedhomeip/pull/38314)
[Adding python3 test module for TC_ACL_2_5 #38180](https://github.com/project-chip/connectedhomeip/pull/38180)
[[Test] Creating ACL 2 6 Python3 Test Module #38380](https://github.com/project-chip/connectedhomeip/pull/38380)
[Adding python3 test module for TC_ACL_2_7 #38173](https://github.com/project-chip/connectedhomeip/pull/38173)
[Create ACL_2_8 python3 test module #38286](https://github.com/project-chip/connectedhomeip/pull/38286)

As this issue refers to their previous yaml versions, please open new issue if necessary.

Closing this issue.

closed 2025-06-29 by raul-marquez-csa