project-chip/connectedhomeip #30193: [Feature] Electrical Power Measurement and Electrical Energy Measurement clusters are not implemented

PSONALl, 2023-11-03:
### Feature description

Electrical Power Measurement and Electrical Energy Measurement clusters are present in the latest spec but the implementation is not there in upstream

Are there any plans to add them?

### Platform

all

jamesharrow, 2023-11-06:
These will be added once the spec reviews have been completed.

HamzaHajeir, 2024-10-08:
Hi there, 

Is there any progress regarding the Electrical Power and Energy Measurements? 

What do you recommend to us to achieve such purposes?

jamesharrow, 2024-10-09:
This issue was implemented in Matter 1.3,

So the code is already there - so this is a stale issue and should be closed.

You can find the example/energy-mamagement-app has a working example in the EVSE and it is also tested in the TC_EPM and TC_EEM test cases in arc/pytjhon_testing folder.

Please check the Matter 1.3 specification for details on how the cluster should be used.

closed 2024-10-09 by jamesharrow