project-chip/connectedhomeip #30194: [1.3] cross-compile for linux kernel 4.9.84 armv7l

tnk1234, 2023-11-03:
### Reproduction steps

I follow matter build guide step by step and build success out/host on my linux PC (ubuntu 22.04, x86_x64).
I want to cross-compile matter for my linux device (linux kernel 4.9.84 armv7l), so i config toolchain:
     declare -x ARCH="arm"
     declare -x CROSS_COMPILE="arm-linux-gnueabihf-"
     gn gen out/unified --args='is_debug=true target_os="linux" target_cpu="arm"'
     ninja -C out/unified
i get error log: 
     ninja: Entering directory `out/unified/'
[428/2239] cc obj/BUILD_DIR/gen/include/platform/Linux/dbus/openthread/openthread.introspect.c.o
FAILED: obj/BUILD_DIR/gen/include/platform/Linux/dbus/openthread/openthread.introspect.c.o 
arm-linux-gnueabihf-gcc -MMD -MF obj/BUILD_DIR/gen/include/platform/Linux/dbus/openthread/openthread.introspect.c.o.d -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wundef -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -Wno-cast-function-type -Wno-psabi -Wno-maybe-uninitialized -fdiagnostics-color -fno-strict-aliasing -Wno-unreachable-code -I/usr/include/gio-unix-2.0 -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -std=gnu11  -Igen/include -c gen/include/platform/Linux/dbus/openthread/introspect.c -o obj/BUILD_DIR/gen/include/platform/Linux/dbus/openthread/openthread.introspect.c.o
In file included from /usr/include/glib-2.0/glib/gtypes.h:32,
                 from /usr/include/glib-2.0/glib/galloca.h:32,
                 from /usr/include/glib-2.0/glib.h:30,
                 from /usr/include/glib-2.0/gobject/gbinding.h:28,
                 from /usr/include/glib-2.0/glib-object.h:22,
                 from /usr/include/glib-2.0/gio/gioenums.h:28,
                 from /usr/include/glib-2.0/gio/giotypes.h:28,
                 from /usr/include/glib-2.0/gio/gio.h:26,
                 from gen/include/platform/Linux/dbus/openthread/introspect.h:12,
                 from gen/include/platform/Linux/dbus/openthread/introspect.c:13:
gen/include/platform/Linux/dbus/openthread/introspect.c: In function ‘openthread_io_openthread_border_router_default_init’:
/usr/lib/x86_64-linux-gnu/glib-2.0/include/glibconfig.h:67:33: error: conversion from ‘long long unsigned int’ to ‘guint64’ {aka ‘long unsigned int’} changes value from ‘18446744073709551615’ to ‘4294967295’ [-Werror=overflow]
 #define G_GUINT64_CONSTANT(val) (val##UL)
                                 ^~~~~~~~~
/usr/include/glib-2.0/glib/gtypes.h:101:21: note: in expansion of macro ‘G_GUINT64_CONSTANT’
 #define G_MAXUINT64 G_GUINT64_CONSTANT(0xffffffffffffffff)
                     ^~~~~~~~~~~~~~~~~~
gen/include/platform/Linux/dbus/openthread/introspect.c:1816:67: note: in expansion of macro ‘G_MAXUINT64’
     g_param_spec_uint64 ("ext-pan-id", "ExtPanId", "ExtPanId", 0, G_MAXUINT64, 0, G_PARAM_READWRITE | G_PARAM_STATIC_STRINGS));
                                                                   ^~~~~~~~~~~
/usr/lib/x86_64-linux-gnu/glib-2.0/include/glibconfig.h:67:33: error: conversion from ‘long long unsigned int’ to ‘guint64’ {aka ‘long unsigned int’} changes value from ‘18446744073709551615’ to ‘4294967295’ [-Werror=overflow]
 #define G_GUINT64_CONSTANT(val) (val##UL)
                                 ^~~~~~~~~
/usr/include/glib-2.0/glib/gtypes.h:101:21: note: in expansion of macro ‘G_GUINT64_CONSTANT’
 #define G_MAXUINT64 G_GUINT64_CONSTANT(0xffffffffffffffff)
                     ^~~~~~~~~~~~~~~~~~
gen/include/platform/Linux/dbus/openthread/introspect.c:1879:87: note: in expansion of macro ‘G_MAXUINT64’
     g_param_spec_uint64 ("extended-address", "ExtendedAddress", "ExtendedAddress", 0, G_MAXUINT64, 0, G_PARAM_READWRITE | G_PARAM_STATIC_STRINGS));
                                                                                       ^~~~~~~~~~~
gen/include/platform/Linux/dbus/openthread/introspect.c: At top level:
cc1: error: unrecognized command line option ‘-Wno-unknown-warning-option’ [-Werror]
cc1: all warnings being treated as errors
[441/2239] ACTION //src/controller/data_model:data_model_zapgen_zap_pregen(//build/toolchain/linux:linux_arm_gcc)
ninja: build stopped: subcommand failed.

### Bug prevalence

alway

### GitHub hash of the SDK that was being used

2b91805ddd2f1c10bf63d1f8d34a409f7517ac29

### Platform

other

### Platform Version(s)

Linux kernel 4.9.84, armv7l

### Type

Test Improvement

rmwiseman, 2024-04-18:
I've also got:

`/usr/lib/x86_64-linux-gnu/glib-2.0/include/glibconfig.h:67:33: error: conversion from ‘long long unsigned int’ to ‘guint64’ {aka ‘long unsigned int’} changes value from ‘18446744073709551615’ to ‘4294967295’ [-Werror=overflow]`

I'm running `gn gen out/armhf --args='target_cpu="arm"'` followed by `ninja -C out/armhf`.  I've just run `sudo apt-get install gcc-arm-linux-gnueabihf` to get past a previous error, but now I get the one above.

Any solution to this...?

andy31415, 2024-10-21:
Some cross compile support exists for rpi builds. 

https://github.com/project-chip/connectedhomeip/blob/master/integrations/docker/images/stage-1/chip-build-crosscompile/README.md#image-description describes how to create a sysroot image for that and maybe you can apply that for your own environment.

Beyond that however, I don't think we can provide SDK developer support for various custom build environments as the combinations are too many.

closed 2024-10-21 by andy31415