project-chip/connectedhomeip #30204: [YAML]

lpbeliveau-silabs, 2023-11-03:
### Workflow issues

In the Yaml CI, we currently need to explicitly test for all members of complex structures even if we are only interested in certain values. For instance, in Test_TC_S_2_2 [here](https://github.com/project-chip/connectedhomeip/blob/78c8704a4e1dacacf0eef824268143c13e122e8c/src/app/tests/suites/certification/Test_TC_S_2_2.yaml#L452C1-L521C24), we are only interested in on Attibute in two clusters, but end up checking all attributes present in the struct, which is tedious and means we will need to modifiy this test each time a new extension field set is implemented in the all-cluster-app:
```
                value:
                    [
                        {
                            ClusterID: 0x0006,
                            AttributeValueList:
                                [{ AttributeID: 0x0000, AttributeValue: 0x01 }],
                        },
                        {
                            ClusterID: 0x0008,
                            AttributeValueList:
                                [{ AttributeID: 0x0000, AttributeValue: 0x64 }],
                        },
                        {
                            ClusterID: 0x0300,
                            AttributeValueList:
                                [
                                    {
                                        AttributeID: 0x0003,
                                        AttributeValue: 0x616b,
                                    },
                                    {
                                        AttributeID: 0x0004,
                                        AttributeValue: 0x607d,
                                    },
                                    {
                                        AttributeID: 0x4000,
                                        AttributeValue: 0x0000,
                                    },
                                    {
                                        AttributeID: 0x0001,
                                        AttributeValue: 0x00,
                                    },
                                    {
                                        AttributeID: 0x4002,
                                        AttributeValue: 0x00,
                                    },
                                    {
                                        AttributeID: 0x4003,
                                        AttributeValue: 0x00,
                                    },
                                    {
                                        AttributeID: 0x4004,
                                        AttributeValue: 0x0019,
                                    },
                                    {
                                        AttributeID: 0x0007,
                                        AttributeValue: 0x00,
                                    },
                                    {
                                        AttributeID: 0x4001,
                                        AttributeValue: 0x02,
                                    },
                                ],
                        },
                    ]
```

As Identified in the TODO below, we need to have a way to check only for the values that we are expecting to see a change in:
```
                #TODO : Need to change the check for a check like below if possible
                # constraints:
                #     type: list
                #     contains: [
                #         {
                #             ClusterID: 0x0006,
                #             AttributeValueList:
                #                 [{ AttributeID: 0x0000, AttributeValue: 0x01 }],
                #         },
                #         {
                #             ClusterID: 0x0008,
                #             AttributeValueList:
                #                 [{ AttributeID: 0x0000, AttributeValue: 0x64 }],
                #         },
                #     ]
```

### Platform

all

bzbarsky-apple, 2023-11-03:
@lpbeliveau-silabs This should be supported ever since #29643 merged a few weeks ago.  Did you try it and it didn't work?

lpbeliveau-silabs, 2023-11-03:
I am currently getting a parsing error when try to run the following step:
`      response:
          values:
              - name: "Status"
                value: 0x00
              - name: "GroupID"
                value: G1
              - name: "SceneID"
                value: 0x01
              - name: "TransitionTime"
                value: 0x0000
              - name: "ExtensionFieldSets"
                constraints:
                    type: list
                    contains: [
                        {
                            ClusterID: 0x0006,
                            AttributeValueList:
                                [{ AttributeID: 0x0000, AttributeValue: 0x01 }],
                        },
                        {
                            ClusterID: 0x0008,
                            AttributeValueList:
                                [{ AttributeID: 0x0000, AttributeValue: 0x64 }],
                        },
                    ]`
                    
 The resulting error is:
 `Parsing: src/app/tests/suites/certification/Test_TC_S_2_2.yaml
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : ✗ 18.0ms
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		Error at step 28:
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		label: 'Step 5: TH sends a ViewScene command to DUT with the GroupID field set to
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		  G1 and the SceneID field set to 0x01.'
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		PICS: S.S.C01.Rsp && PICS_SDK_CI_ONLY
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		command: ViewScene
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		arguments:
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		  values:
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		  - name: GroupID
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		    value: G1
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		  - name: SceneID
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		    value: 1
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		response:
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		  values:
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		  - name: Status
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		    value: 0
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		  - name: GroupID
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		    value: G1
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		  - name: SceneID
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		    value: 1
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		  - name: TransitionTime
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		    value: 0
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		  - name: ExtensionFieldSets
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		    constraints:
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		      type: list
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		      contains:
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		      - ClusterID: 6 <-- Unknown key "ClusterID"
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		        AttributeValueList:
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		        - AttributeID: 0
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		          AttributeValue: 1
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		      - ClusterID: 8
2023-11-03 16:17:25.196 ERROR   16:17:25.041 - TEST OUT  : 		        AttributeValueList:
2023-11-03 16:17:25.197 ERROR   16:17:25.041 - TEST OUT  : 		        - AttributeID: 0
2023-11-03 16:17:25.197 ERROR   16:17:25.041 - TEST OUT  : 		          AttributeValue: 100`

Seems to indicate that the parser is not expecting this key here but this same key works when used to check for a specific value.

bzbarsky-apple, 2023-11-04:
Yep, there's a bug in the handling of `contains` for command responses like this.  Fix coming up.

closed 2023-11-06 by lpbeliveau-silabs