project-chip/connectedhomeip #30221: [BUG] Flake on Darwin TestReliableMessageProtocol

cecille, 2023-11-05:
### Reproduction steps

Please see https://csamembers.slack.com/archives/G014G30SVV0/p1699029897819219. The flake rate of this test seems to be on the rise. Given that this is related to reliable messaging, there is some concern that this might be related to the ICD work and might be a canary indicating a timing bug of some type.

### Bug prevalence

Unsure - regular, but I can't wager a guess as to prevalence

### GitHub hash of the SDK that was being used

b82bea21749dfbf6c837c668bc2d1bac2df9da94

### Platform

darwin

ksperling-apple, 2023-11-09:
Just filed this dupe before being pointed to this issue: https://github.com/project-chip/connectedhomeip/issues/30395
There are a couple of CI runs referenced there in case that helps. Will close the dupe.

kiel-apple, 2024-11-02:
maybe flaking again? https://github.com/project-chip/connectedhomeip/actions/runs/11636681494/job/32408518189?pr=36343