project-chip/connectedhomeip #30223: [1.1] I want to make chip-tool as a OTA provider. What would be chip-tool node ID?

Shahmeera, 2023-11-06:
### Reproduction steps

For generating chip-tool as a OTA provider I have enabled OTA-provider cluster as a server from ZAP tool. So the process would be like below
(1) Commission OTA requestor from chip-tool
(2) Announce OTA provider(chip-tool) for presence. For this step I wanted to know chip-tool node ID
(3) According to my knowledge if only chip-tool and ota requestor are involved, there would be no role of access control list(ACL) , am I right?

### Bug prevalence

none

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip/tree/v1.2.0.1

### Platform

IMX8

### Type

Test Improvement

bzbarsky-apple, 2023-11-07:
```
chip-tool pairing get-commissioner-node-id 
```

and look for a log like so:

```
[1699383064125] [20440:24284350] [TOO] Commissioner Node Id 0x:000000000001B669
```

closed 2023-11-07 by bzbarsky-apple

Shahmeera, 2023-11-08:
Thank you for your answer. Now I am facing only one error here in the function **ProcessCommandDataIB** (/src/app/CommandHandler.cpp). Here I am getting error log like this "acl err code is src/access/AccessControl.cpp:314: CHIP Error 0x00000003: Incorrect state"

`VerifyOrExit(mExchangeCtx && mExchangeCtx->HasSessionHandle(), err = CHIP_ERROR_INCORRECT_STATE);

    {   
        ChipLogDetail(DataManagement,"cluster %x and endpoint %d\n",concretePath.mClusterId,concretePath.mEndpointId);
        Access::SubjectDescriptor subjectDescriptor = GetSubjectDescriptor();
        Access::RequestPath requestPath{ .cluster = concretePath.mClusterId, .endpoint = concretePath.mEndpointId };
        Access::Privilege requestPrivilege = RequiredPrivilege::ForInvokeCommand(concretePath);
        err                                = Access::GetAccessControl().Check(subjectDescriptor, requestPath, requestPrivilege);
        ChipLogError(DataManagement, "acl err code is %" CHIP_ERROR_FORMAT, err.Format());
        if (err != CHIP_NO_ERROR)
        {
            if (err != CHIP_ERROR_ACCESS_DENIED)
            {
               // ChipLogDetail(DataManagement,"[MS::: failed concrete path %s] \n",concretePath);
                return FallibleAddStatus(concretePath, Status::Failure) != CHIP_NO_ERROR ? Status::Failure : Status::Success;
            }
             ChipLogDetail(DataManagement,"TODO wildcard\n");
            // TODO: when wildcard invokes are supported, handle them to discard rather than fail with status
            return FallibleAddStatus(concretePath, Status::UnsupportedAccess) != CHIP_NO_ERROR ? Status::Failure : Status::Success;
        } 
    }`
    why this is showing incorrect state, as we are as a matter root node so it has all the privilege?