project-chip/connectedhomeip #30237: [Question] ESP32 device type recognition HomeApp

BartJozwicki, 2023-11-06:
### Reproduction steps

I am not sure where to post this questions so sorry if I labelled it incorrectly. 

I am building a smart light using ESP32-H2, I have my code working based on light-app example, I can control brightness of the light using sliders in the HomeAPP (iOS) but I would like to have only one slider present in the HomeApp rather than 6 different colours from the RGBW example. Could you please point me to the parts of code or some documentation where I can solve that? I suppose there is some kind of identifier in the code that sends information to the HomeApp but I have a hard time finding it. Thank you for help. 

### Platform

esp32

### Type

Platform validated

dhrishi, 2023-11-06:
@BartJozwicki Can you please provide the details about the example you are using, is it the lighting-app? Also, any screenshots from the Home app would help

BartJozwicki, 2023-11-06:
> @BartJozwicki Can you please provide the details about the example you are using, is it the lighting-app? Also, any screenshots from the Home app would help

I am using Lighting-app as my template yes.

Sure that is the view from my HomeApp:

[image]

My goal is to have only one of the sliders for brightness (at the moment you can see 6 different colours)

bzbarsky-apple, 2023-11-07:
@BartJozwick Make sure the device type does not say "color light" on the endpoint in ZAP, and disable the color control cluster on that endpoint.

closed 2023-11-07 by bzbarsky-apple

BartJozwicki, 2023-11-08:
> @BartJozwick Make sure the device type does not say "color light" on the endpoint in ZAP, and disable the color control cluster on that endpoint.

I have changed the lighting-app.zap as suggested:
(enable color control: 0)

[image]

Device says Dimmible Light:

[image]

After changes I have deleted build files in the lighting-app and used idf.py build
I get the following errors when building the file:

[image]

Could you please advise what is wrong ?

bzbarsky-apple, 2023-11-14:
See https://github.com/project-chip/connectedhomeip/issues/30324#issuecomment-1802270280