project-chip/connectedhomeip #30238: [1.1] [NXP] [K32W]

gokul-2000, 2023-11-06:
### Reproduction steps

Hi, 
We attempted to create a custom BLE UUID implementation. To achieve this, we modified two files: gatt_db.h and gatt_uuid128.h. We added the custom UUID using a certain method.

Changes in **gatt_db.h** file.
```
PRIMARY_SERVICE_UUID128(service_wireless, uuid_service_wireless)
    CHARACTERISTIC_UUID128(char_thread_network_info, uuid_thread_network_info, (gGattCharPropWrite_c))
        VALUE_UUID128_VARLEN(value_thread_network_info, uuid_thread_network_info, (gPermissionFlagWritable_c), gAttMaxWriteDataSize_d(gAttMaxMtu_c), 1, 0x00)
    CHARACTERISTIC_UUID128(char_end_device_info, uuid_end_device_info, (gGattCharPropRead_c))
        VALUE_UUID128_VARLEN(value_end_device_info, uuid_end_device_info, (gPermissionFlagReadable_c), gAttMaxReadDataSize_d(gAttMaxMtu_c), 9, "undefined")
    CHARACTERISTIC_UUID128(char_end_device_reboot, uuid_end_device_reboot, (gGattCharPropWrite_c))
        VALUE_UUID128_VARLEN(value_end_device_reboot, uuid_end_device_reboot, (gPermissionFlagWritable_c), gAttMaxReadDataSize_d(gAttMaxMtu_c), 1, 0x00)
```
Changes in **gatt_uuid128.h** file.
```
UUID128(uuid_service_wireless, 0x93, 0x3d, 0x87, 0xbd, 0x63, 0xd6, 0x10, 0xb2, 0xe9, 0x11, 0xa4, 0x2b, 0x46, 0x01, 0xb2, 0x14)
UUID128(uuid_thread_network_info, 0x93, 0x3d, 0x87, 0xbd, 0x63, 0xd6, 0x10, 0xb2, 0xe9, 0x11, 0xa4, 0x2b, 0x9e, 0x03, 0xb2, 0x14)
UUID128(uuid_end_device_info, 0x93, 0x3d, 0x87, 0xbd, 0x63, 0xd6, 0x10, 0xb2, 0xe9, 0x11, 0xa4, 0x2b, 0xe8, 0x04, 0xb2, 0x14)
UUID128(uuid_end_device_reboot, 0x93, 0x3d, 0x87, 0xbd, 0x63, 0xd6, 0x10, 0xb2, 0xe9, 0x11, 0xa4, 0x2b, 0xc6, 0x08, 0xb2, 0x14)
```

We have encountered an issue with obtaining UUID in BLE after the connection. Although we have attempted to perform some read and write operations, we have not received any logs in **blekw_gatt_server_cb** in _gEvtAttributeRead_c_ and _gEvtAttributeWritten_c_ events.

To address this issue, we have registered the UUID in both advertisement data and scan response data. Please refer to the diff file for further details.
[lighting_app_diff.txt](https://github.com/project-chip/connectedhomeip/files/13267474/lighting_app_diff.txt)

Is this the correct way to add a custom UUID as the above implementation? If not then please guide us!!

Thank you in advance for your assistance.

### Bug prevalence

Everytime

### GitHub hash of the SDK that was being used

v1.1.0.1-tag-nxp

### Platform

k32w

### Platform Version(s)

1.1

### Type

Core SDK Crash

bzbarsky-apple, 2023-11-07:
No idea about NXP bits.  @doru91 might know something.

gokul-bhalodiya-7, 2023-11-07:
Thanks for the response @bzbarsky-apple 

@ bzbarsky-apple I would like to know if we are developing the matter end-device from the CHIP stack then we can add a new BLE UUID or not to fulfill extra feature needs.

Any help would be precious @doru91 , Regarding BLE UUID attribute read, write events not coming in the GATT server callback(blekw_gatt_server_cb).

bzbarsky-apple, 2023-11-14:
I believe you can generally do whatever over BLE that's not Matter-related.

gokul-bhalodiya-7, 2023-11-14:
Thank you very much for the reply @bzbarsky-apple 

Adding UUID is getting failed due to improper indentation(due to mismatch in space and tabs) in **gatt_db.h** file

You can close this issue.

closed 2023-11-14 by bzbarsky-apple