project-chip/connectedhomeip #30273: [1.3] The implementation of the Demand Response Load Control Cluster is required

Rajashreekalmane, 2023-11-07:
### Reproduction steps

1. Executed the below mentioned command to put DUT into a commissionable state
./chip-all-clusters-app

2. Once DUT reach the commissionable state sent below mentioned command on TH.
./chip-tool pairing onnetwork 1 20202021

3. After the successful commissioning, I attempted to validate the Demand Response Load Control Cluster, but it is not implemented in chip-tool
[image]

### Bug prevalence

  Whenever I do

### GitHub hash of the SDK that was being used

24d38dac1000c0e57143a2a481268ecbcd398138

### Platform

raspi

### Type

Test Improvement

### Anything else?

**Spec Reference :** 
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/DemandResponseLoadControl.adoc

jamesharrow, 2023-11-24:
Believe that this is now in the SDK.
@bzbarsky-apple ?

bzbarsky-apple, 2023-12-06:
Yes, chip-tool has this now.

closed 2023-12-06 by bzbarsky-apple