project-chip/connectedhomeip #30275: [1.2] Commissioning issues with chip-tool with Linux based lighting-app wifi.

Rajesh0071998, 2023-11-07:
### Reproduction steps

Build the Lighting-app for the linux and running on the ubuntu machine and having a chip-tool on other linux machine.
But while commissioning I think issues is related to operational discovery.

Logs :
[Device_logs.txt](https://github.com/project-chip/connectedhomeip/files/13279219/Device_logs.txt)
[chip-tool.txt](https://github.com/project-chip/connectedhomeip/files/13279225/chip-tool.txt)

How to resolve the above issue.

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

.

### Platform

other

### Type

Platform Issue

bzbarsky-apple, 2023-11-07:
Commissioner is commissioning node id 2, but device never advertises that (it advertises node id 9 and node id 19).  The AddNOC in the device log is definitely for node id 9.

So either something is _very_ broken here, or these logs don't have anything to do with each other.

Rajesh0071998, 2023-11-08:
Oohoo i am so sorry i have posted the worng one . below are the logs of Device and chip-tool.
[Logs_device.txt](https://github.com/project-chip/connectedhomeip/files/13292087/Logs_device.txt)
[chip-tool.txt](https://github.com/project-chip/connectedhomeip/files/13292097/chip-tool.txt)

I am using mobile hotspot as Access Point.

Rajesh0071998, 2023-11-09:
Is it necessary for a WiFi device to be connected to WiFi or Ethernet before commissioning on same network as chip-tool is on?

bzbarsky-apple, 2023-11-14:
> Is it necessary for a WiFi device to be connected to WiFi or Ethernet before commissioning on same network as chip-tool is on?

If it's wifi-only and has not BLE, then yes.

bzbarsky-apple, 2023-11-14:
Those last logs look like the device not joining "the network" properly.... where "the network" should be the network chip-tool is on.

Rajesh0071998, 2023-11-14:
As i am using Mobile hotspot. do i have to change the network?

Rajesh0071998, 2023-11-14:
After the commissioning fails after some time device joins the network and then pairing is been done.

Rajesh0071998, 2023-11-14:
any idea how to resolve this issue.

bzbarsky-apple, 2023-11-15:
What is the issue?  That the device takes longer to join the network than the commissioner waits for it?

What does the device claim about the time it will take it to join networks?  How does that compare to the actual time it takes?

Rajesh0071998, 2023-11-21:
Looks like the device is taking time to join the network and mean while the chip_tool closed it's commissioning window.

Attaching the logs contain the fails and success logs.
[chip_tool_logs.txt](https://github.com/project-chip/connectedhomeip/files/13422059/chip_tool_logs.txt)
[device_logs.txt](https://github.com/project-chip/connectedhomeip/files/13422062/device_logs.txt)

Rajesh0071998, 2023-11-24:
Initially when we start the commissioning and run on other terminal -> avahi-browse -a
IT doesn't show Matter operational advertisements (_matter._tcp).
when it fails and we wait for the device to come back to its commissioning stage.Now when we start recommissioning Now it shows the (_matter._tcp) advertisments  and commissioning gets success full.

bzbarsky-apple, 2023-12-06:
Sounds like a buggy device, if it's responding "success" to the "join a network" command before it's actually done that?

rlux-bluezone, 2024-02-28:
I experienced the same bug.

On the initial commissioning, the app gets to the point of connecting to the Wi-Fi and starting the operational Advertisement. But the Device is not able to advertise on the Wi-Fi interface because it isn't marked as active (connected to the Wi-Fi), when the advertisement is started.

On the second try the Device is already connected to the Wi-Fi (interface is marked as active) so there is no Problem to start the operational advertisement.

For me the fix to the issue is to add
```C++
app::DnssdServer::Instance().StartServer();
```
on line 82 in "src\app\server\CommissioningWindowManager.cpp".

This "reloads" the active Interfaces, so the operational advertisement is started on the Wi-Fi interface. Then BLE commissioning should work now.

It also looks like the "NetworkCommissioning::Status::kSuccess" Response is sent prematurely in "\src\platform\Linux\ConnectivityManagerImpl.cpp" on Line 489, when the check for assigned addresses is done in PostNetworkConnect() (line 492) after the Response is sent.

rlux-bluezone, 2024-03-05:
No. "OperationalNetworkEnabled"

bzbarsky-apple, 2024-03-05:
@rlux-bluezone line numbers should come along with a SHA, otherwise people end up looking at different code that happens to be on the same line.

So you're saying changing the AdvertiseOperational() there to StartServer() makes things work?

rlux-bluezone, 2024-03-06:
Sorry for the confusion(im on [https://github.com/project-chip/connectedhomeip/commit/181b0cb14ff007ec912f2ba6627e05dfb066c008](https://github.com/project-chip/connectedhomeip/commit/181b0cb14ff007ec912f2ba6627e05dfb066c008).) 

I added StartServer() above the AdvertiseOperational() call (i guess just changing it will work as well because StartServer calls AdvertiseOperational()). After that BLE-commissioning works again.

If i understand it correctly, then the interfaces are added in the Listen() function in "src/lib/dnssd/minimal_mdns/Server.cpp" on Line 248 with the mEndpoints.CreateObject(...) call. 

Listen(...) is called via StartServer() but not in AdvertiseOperational().

bzbarsky-apple, 2024-03-12:
@rlux-bluezone Thank you for digging through that.

@andy31415 should probably think about the minimal mdns case here, but he's out until Monday next week...

andy31415, 2024-03-18:
My understanding of expectations was that devices should listen for changes to connectivity and restart relevant events. For example https://github.com/project-chip/connectedhomeip/blob/master/examples/chef/esp32/main/main.cpp#L75

although a quick grep for kInternetConnectivityChange seems to only show some subset of examples doing that.

bzbarsky-apple, 2024-03-19:
@andy31415 Well, what are the semantics around kInternetConnectivityChange vs kOperationalNetworkEnabled (not to be confused with kOperationalNetworkStarted, apparently)?  

There is no documentation for kOperationalNetworkEnabled.  And "communicate with the Internet" (documentation for kInternetConnectivityChange) has nothing to do with Matter operation, right?  Matter devices should work fine without being able to "communicate with the internet".

Also, right now `kInternetConnectivityChange` is dispatched for things like "we are on Linux and have an IPv4 address" (but not IPv6, note), so it clearly seems like the wrong thing....

bzbarsky-apple, 2024-03-19:
But the bigger question, @andy31415 is that the DNS-SD APIs need to clearly define when you need to "restart" vs just "AdvertiseOperational".  The fact that AdvertiseOperational is the wrong thing to use for changes to the set of network interfaces is _not_ documented in the API, and seems like a quirk of the min-mdns implementation....

andy31415, 2024-03-20:
what is happening is that the https://github.com/project-chip/connectedhomeip/blob/master/src/lib/dnssd/MinimalMdnsServer.cpp#L67 in StartServer is what passes in a list of endpoints to listen on. So anything that changes that list needs to re-start the server.

Currently the SDK does not seem to have a mechanism to determine what changes the available endpoints. Some examples use `kInternetConnectivityChange` (which seems to be more IP changed than actual internet ... not sure) however overall minmdns does not keep track of networking state and I am not sure we have documented way to watch for the state of inet.

andy31415, 2024-03-20:
Just interface IDs are not enough either: https://github.com/project-chip/connectedhomeip/blob/master/src/lib/dnssd/minimal_mdns/AddressPolicy_DefaultImpl.cpp#L57 actually checks if an interface has an IP address assigned, so if Ip addresses are updated on interfaces, restart is also required for minmdns (but not for avahi/bonjour and such since those likely have O/S support to detect interface changes).

bzbarsky-apple, 2024-03-26:
> So anything that changes that list needs to re-start the server.

Absolutely none of that is documented in the public dns-sd framework APIs.