project-chip/connectedhomeip #30295: Question of NetworkCommissioning featureMap implementation

joonhaengHeo, 2023-11-08:
### Reproduction steps

I would like to inquire because there seems to be a difference between the SDK implementation and the core specification.

https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/service_device_management/NetworkCommissioningCluster.adoc#4-features
Looking at the Spec above, it is understood that Features can support Wi-Fi, Thread, and Ethernet all.

However, looking at the code implementation details below, it seems that only one can be enabled, so I am curious which one is correct.
https://github.com/project-chip/connectedhomeip/blob/155917e529db5e1e8d7819c6309115fcabc93cff/src/controller/CHIPDeviceController.cpp#L1969

I think one of the two needs to be modified.

### Bug prevalence

everytime

### GitHub hash of the SDK that was being used

155917e529db5e1e8d7819c6309115fcabc93cff

### Platform

core

### Platform Version(s)

master

yunhanw-google, 2023-11-08:
for network commissiong cluster, i remember wifi network cluster is in endpoint0, thread is in endpoint1..different network commissioning cluster lives in different endpoint

joonhaengHeo, 2023-11-08:
I know that when the Commissioner reads, it reads all endpoints.
https://github.com/project-chip/connectedhomeip/blob/155917e529db5e1e8d7819c6309115fcabc93cff/src/controller/CHIPDeviceController.cpp#L2447

However, I could not find a sentence that states that only one network information must be configured for one endpoint. and. The FeatureMap is also divided by bit so that multiple network information can be set.

bzbarsky-apple, 2023-11-08:
@joonhaengHeo The "Conformance" column for the features in question in the spec says `O.a` for all of them.  That means "exactly one of these is allowed".  See https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/Data-Model.adoc#313-choice

bzbarsky-apple, 2023-11-08:
Also:

>  I could not find a sentence that states that only one network information must be configured for one endpoint. 

This is a general restriction: you can't have two instances of the same cluster ID on the same endpoint, because there is no way to address them, right?

joonhaengHeo, 2023-11-08:
@bzbarsky-apple 
What does it mean that a Network Commissioning Cluster must have two instances? Of course, depending on the device implementation, it may have to be divided into two instances, but I understand that supporting more than two features with one instance is not a violation of the specification. In particular, I know that other features will be added after Matter 1.3, but I don't think there should be such restrictions in the controller.

joonhaengHeo, 2023-11-10:
I misunderstand. Thank you for response.

closed 2023-11-10 by joonhaengHeo

bzbarsky-apple, 2023-11-14:
> What does it mean that a Network Commissioning Cluster must have two instances?

Only if you support two different network interfaces...

> but I understand that supporting more than two features with one instance is not a violation of the specification

It is if the features are defined to be mutually exclusive.