project-chip/connectedhomeip #30298: [1.2] I want to make chip-tool as an OTA provider. But I am facing error related to ACL.

Shahmeera, 2023-11-08:
### Reproduction steps

For generating chip-tool as a OTA provider I have enabled OTA-provider cluster as a server from ZAP tool. So the process would be like below
(1) Commission OTA requestor from chip-tool
(2) Announce OTA provider(chip-tool) for presence. For this step I knew chip-tool node ID would be 112233.
(3) According to my knowledge if only chip-tool and ota requestor are involved, there would be no role of access control list(ACL) , am I right?
Now I am facing only one error here in the function ProcessCommandDataIB (/src/app/CommandHandler.cpp). Here I am getting error log like this "acl err code is src/access/AccessControl.cpp:314: CHIP Error 0x00000003: Incorrect state"

`  {   
    ChipLogDetail(DataManagement,"cluster %x and endpoint %d\n",concretePath.mClusterId,concretePath.mEndpointId);
    Access::SubjectDescriptor subjectDescriptor = GetSubjectDescriptor();
    Access::RequestPath requestPath{ .cluster = concretePath.mClusterId, .endpoint = concretePath.mEndpointId };
    Access::Privilege requestPrivilege = RequiredPrivilege::ForInvokeCommand(concretePath);
    err                                = Access::GetAccessControl().Check(subjectDescriptor, requestPath, requestPrivilege);
    ChipLogError(DataManagement, "acl err code is %" CHIP_ERROR_FORMAT, err.Format());
    if (err != CHIP_NO_ERROR)
    {
        if (err != CHIP_ERROR_ACCESS_DENIED)
        {
           // ChipLogDetail(DataManagement,"[MS::: failed concrete path %s] \n",concretePath);
            return FallibleAddStatus(concretePath, Status::Failure) != CHIP_NO_ERROR ? Status::Failure : Status::Success;
        }
         ChipLogDetail(DataManagement,"TODO wildcard\n");
        // TODO: when wildcard invokes are supported, handle them to discard rather than fail with status
        return FallibleAddStatus(concretePath, Status::UnsupportedAccess) != CHIP_NO_ERROR ? Status::Failure : Status::Success;
    } 
}`
why this is showing incorrect state, as we are as a matter root node so it has all the privilege?

### Bug prevalence

none

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip/tree/v1.2.0.1

### Platform

IMX8

### Type

Test Improvement

bzbarsky-apple, 2023-11-08:
> According to my knowledge if only chip-tool and ota requestor are involved, there would be no role of access control list

That's not correct.  ACLs are always enforced for server clusters.  For chip-tool there is no useful access control implementation set up, apparently, so the access control checks error out and "fail closed" (that is, no access allowed if there is an error checking access).

If you want chip-tool to support server clusters, you will need to fix it to provide an access control implementation.

closed 2023-11-08 by bzbarsky-apple

Shahmeera, 2023-11-10:
./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"fabricIndex": 1, "privilege": 3, "authMode": 2, "subjects": null, "targets": [{"cluster": 41, "endpoint": null, "deviceType": null}]}]' <provider node id> 0
I want to assign privileges to the chip-tool in my case(chip-tool+ota-provider) 112233. How can I do that?

bzbarsky-apple, 2023-11-14:
"write acl" will write ACLs to the other device.  You are trying to set up ACLs on chip-tool.  It does not have a way to do that right now; you will need code changes for that.