project-chip/connectedhomeip #30299: [1.3] Actions Cluster XML has to be updated as per spec

KishokG, 2023-11-08:
### Reproduction steps

**XML has to be updated for below attribute:**

- **ACT.S.A0001(EndpointLists) -** Attribute name has to be updated from EndpointList to EndpointLists.

**XML Reference:**

[image]

**Specification Reference:**

[image]

### Bug prevalence

Whenever I do this 

### GitHub hash of the SDK that was being used

eb2f8f6ca95c181577415bc4b3e7cc240d5b2edd

### Platform

raspi

### Platform Version(s)

RPI - 4, 8GB RAM

### Type

Test Improvement

bzbarsky-apple, 2023-11-08:
The `define` does not matter. The name of the attribute in that XML is correct.

closed 2023-11-08 by bzbarsky-apple

KishokG, 2023-11-08:
Yeah got it. Thank you @bzbarsky-apple