project-chip/connectedhomeip #30300: [1.3] Bridged Device Basic Information Cluster XML has to be updated as per spec

KishokG, 2023-11-08:
### Reproduction steps

**XML has to be updated for below event:**

- **BRBINFO.S.E02(Leave) -** FabricIndex field has to be added for the Leave event.

**XML Reference:**

[image]

**Specification Reference:**

[image]

**Note:** Bridged Device Basic Information Cluster is derived from Basic Information Cluster.

### Bug prevalence

Whenever I do this 

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip/commit/eb2f8f6ca95c181577415bc4b3e7cc240d5b2edd

### Platform

raspi

### Platform Version(s)

RPI - 4, 8GB RAM

### Type

Test Improvement

bzbarsky-apple, 2023-11-08:
Looks like https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/5264 was never followed with an SDK issue.

bzbarsky-apple, 2023-11-08:
Looks like Basic Information as fixed, so this is just Bridged Device Basic Information.