project-chip/connectedhomeip #30306: [1.3] Window Covering  Cluster XML has to be updated as per spec

Saravana-kr22, 2023-11-08:
### Reproduction steps

### Description:

As per the spec the below mentioned Attribute/Command are Non-Matter , but they are availble in the XML, Hence it should be removed from the xml.

- [ ] GoToLiftValue
- [ ] GoToLiftPercentage
- [ ] GoToTiltValue
- [ ] GoToTiltPercentage
- [ ] InstalledOpenLimitLift
- [ ] InstalledClosedLimitLift
- [ ] InstalledOpenLimitTilt
- [ ] InstalledClosedLimitTilt
- [ ] PhysicalClosedLimitLift
- [ ] PhysicalClosedLimitTilt
- [ ] CurrentPositionLift1
- [ ] CurrentPositionTilt1

**XML Reference:** https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/window-covering.xml

**Spec Reference:** https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/WindowCovering.adoc

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

eb2f8f6ca95c181577415bc4b3e7cc240d5b2edd

### Platform

raspi

### Platform Version(s)

RPI - 4, 8GB RAM

### Type

Test Improvement

bzbarsky-apple, 2023-11-08:
> As per the spec the below mentioned Attribute/Command are Non-Matter 

I don't think that's correct.  GoToLiftValue is absolutely part of Matter.  I have no idea why you think it's not @Saravana-kr22 .  Can you please double-check things here?

Saravana-kr22, 2023-11-08:
@bzbarsky-apple As per my understanding the zigbee feature are Non-Matter and in spec the conformance for GoToLiftValue is [LF & ABS], Since the ABS (Absolute position) is for Zigbee. So this is an Non-Matter Command.

Please let me know if my understanding is wrong.

bzbarsky-apple, 2023-11-14:
> Since the ABS (Absolute position) is for Zigbee.

Ah, interesting.  Well, the SDK probably can't remove these things, so this may need a spec fix.  @jmeg-sfy can you check, please, whether the intent was that ABS is never supported in Matter?

jmeg-sfy, 2023-11-15:
@bzbarsky-apple @Saravana-kr22 those were legacy ZigBee, we decided to set those aside for the first versions of Matter and reinstate them if they were needed later. So we could remove those for now the spec is correct , SDK should disable those attributes

bzbarsky-apple, 2023-11-17:
@jmeg-sfy The problem is: they have shipped as public SDK APIs.  Removing them at this point is going to be a years-long deprecation process, etc...

Emill, 2023-12-29:
In the spec we have the following fields for the GoToLiftPercentage command:
0: LiftPercentageValue
1: LiftPercent100thsValue

But in the SDK we have the following fields for the same command (window-covering.xml):
0: liftPercent100thsValue (note that it starts with lowercase `l`)

(so they are not consistent, neither with the casing nor the ids)
Same issue with GoToTiltPercentage.

I came across these during the test TC-WNCV-7.1 in the Matter 1.2 Application Clusters Test Plan. The expected outcome is "Parameter value is 5000 Percent100ths". I am unsure what behaviour to accept/reject and if I must modify the SDK to pass the test. I guess the TH follows the SDK and not the spec...

bzbarsky-apple, 2024-01-02:
The TH uses the SDK.  

The incorrect ID in the SDK is a problem for sure, but not one that can be changed in the SDK without breaking compat...  It needs to be fixed in the spec, sadly.  Note that the two fields there are mutually exclusive, so having them both use id 0 would not technically be a problem.

The casing we should fix in the SDK.  It doesn't affect anything in practice.

Emill, 2024-09-05:
I can see that the spec has now been updated so that LiftPercent100thsValue is now field id 0 instead of 1. However, the .xml file has incorrect casing (liftPercent100thsValue rather than LiftPercent100thsValue etc.). Can this be fixed?

jmeg-sfy, 2024-09-19:
Would be nice to fix the Casing indeed