project-chip/connectedhomeip #30321: [BUG] Conformance issues on light app

cecille, 2023-11-08:
### Reproduction steps

[MatterTest] 11-08 10:22:19.272 INFO 
Problem: ProblemSeverity.ERROR
    test_name: test_TC_IDM_10_2
    location: 
        Endpoint: 0,
        Cluster:  44 (0x2c) TimeFormatLocalization,
        Attribute:1 (0x01)
    problem: Attribute 0x01 is included, but is disallowed by conformance. Conformance: CALFMT, implemented features: 
    spec_location: 

[MatterTest] 11-08 10:22:19.272 INFO 
Problem: ProblemSeverity.ERROR
    test_name: test_TC_IDM_10_2
    location: 
        Endpoint: 0,
        Cluster:  44 (0x2c) TimeFormatLocalization,
        Attribute:2 (0x02)
    problem: Attribute 0x02 is included, but is disallowed by conformance. Conformance: CALFMT, implemented features: 
    spec_location: 

[MatterTest] 11-08 10:22:19.272 INFO 
Problem: ProblemSeverity.ERROR
    test_name: check_spec_conformance_for_commands
    location: 
        Endpoint: 0,
        Cluster:  60 (0x3c) AdministratorCommissioning,
        Command:  1 (0x01)
    problem: Command 0x01 is included, but disallowed by conformance. Conformance: BC, implemented features: 
    spec_location: 

[MatterTest] 11-08 10:22:19.272 INFO 
Problem: ProblemSeverity.ERROR
    test_name: test_TC_IDM_10_2
    location: 
        Endpoint: 1,
        Cluster:  5 (0x05) Scenes,
        Attribute:65532 (0xfffc)
    problem: Required feature with mask 0x04 is not present in feature map. Conformance: M, implemented features: SN,EX,TS
    spec_location: 

[MatterTest] 11-08 10:22:19.272 INFO 
Problem: ProblemSeverity.ERROR
    test_name: test_TC_IDM_10_2
    location: 
        Endpoint: 1,
        Cluster:  5 (0x05) Scenes,
        Attribute:7 (0x07)
    problem: Attribute 0x07 is included, but is disallowed by conformance. Conformance: FS, implemented features: SN,EX,TS
    spec_location: 

[MatterTest] 11-08 10:22:19.272 INFO 
Problem: ProblemSeverity.ERROR
    test_name: test_TC_IDM_10_2
    location: 
        Endpoint: 1,
        Cluster:  8 (0x08) LevelControl,
        Attribute:4 (0x04)
    problem: Attribute 0x04 is included, but is disallowed by conformance. Conformance: FQ, implemented features: OO,LT
    spec_location: 

[MatterTest] 11-08 10:22:19.273 INFO 
Problem: ProblemSeverity.ERROR
    test_name: test_TC_IDM_10_2
    location: 
        Endpoint: 1,
        Cluster:  8 (0x08) LevelControl,
        Attribute:5 (0x05)
    problem: Attribute 0x05 is included, but is disallowed by conformance. Conformance: FQ, implemented features: OO,LT
    spec_location: 

[MatterTest] 11-08 10:22:19.273 INFO 
Problem: ProblemSeverity.ERROR
    test_name: test_TC_IDM_10_2
    location: 
        Endpoint: 1,
        Cluster:  8 (0x08) LevelControl,
        Attribute:6 (0x06)
    problem: Attribute 0x06 is included, but is disallowed by conformance. Conformance: FQ, implemented features: OO,LT
    spec_location: 

### Bug prevalence

always

### GitHub hash of the SDK that was being used

3f91ae9be48c0db144c506f832099d942f33bf0a

### Platform

other

luis4a0, 2025-01-03:
Since there is no script to test only IDM_10.2, the bug can be reproduced locally with the following command:
```
./scripts/tests/run_python_test.py --factory-reset --app ./out/linux-x64-light-no-ble/chip-lighting-app --app-args "--trace-to json:log" --script src/python_testing/TC_DeviceConformance.py --script-args "--commissioning-method on-network --qr-code MT:-24J0AFN00KA0648G00"
```

andy31415, 2025-01-06:
Assigned to myself as a "worked on" placeholder pending getting seats for new developers in the project

closed 2025-01-30 by khodya