project-chip/connectedhomeip #30346: [1.3] Network Commissioning Cluster XML has to be updated as per spec

KishokG, 2023-11-09:
### Reproduction steps

**XML has to be updated as per spec:**

- ClusterRevision value has to be added.
- MaxNetworks attribute min value has to be added.
- In ScanNetworkResponse command, the string length has to be added for DebugText argument.
- In SupportedWiFiBands, SupportedThreadFeatures, and ThreadVersion attributes the XML says apiMaturity="provisional". But in spec its not available in anywhere. Why it is been listed in XML? 

**XML Reference:**
https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/network-commissioning-cluster.xml

**Specification Reference:**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/service_device_management/NetworkCommissioningCluster.adoc

### Bug prevalence

Whenever I do this 

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip/commit/eb2f8f6ca95c181577415bc4b3e7cc240d5b2edd

### Platform

raspi

### Type

Test Improvement

bzbarsky-apple, 2023-11-09:
>  But in spec its not available in anywhere

I am not sure what this means @KishokG

cjandhyala, 2023-11-09:
@bzbarsky-apple He meant, its not mentioned as Provisional in the Spec, but its mentioned as Provisional in the XML, Right @KishokG

KishokG, 2023-11-10:
Yes correct @cjandhyala

bzbarsky-apple, 2023-11-14:
It can't be mentioned as "provisional" in the spec because it's not far enough yet in the spec to be "provisional".  "Provisional" in the spec means "balloted but not certifiable", but none of the 1.3 stuff was even balloted at the point when this issue was filed @cjandhyala @KishokG .  So in the spec it's "even less stable than things that are provisional".

bzbarsky-apple, 2023-11-14:
For purposes of comparing to XML, anything that is not:

1. In a finalized and released spec.
2. And not marked provisional.

should be marked provisional in the XML.