project-chip/connectedhomeip #30348: [1.3] Application Basic Cluster XML has to be updated as per spec

Rajashreekalmane, 2023-11-09:
### Reproduction steps

- [ ]  Need remove 'default=""0x0""' from the XML since the spec does not specify any default value for ProductID and vendor-id
[image]
[image]

- [ ]  AllowedVendorList type should be updated by removing the length=""32""
[image]

[image]

- [ ]  Global attributes are missing in the XML

### Bug prevalence

whenver i do

### GitHub hash of the SDK that was being used

1234567

### Platform

raspi

### Type

Test Improvement

### Anything else?

**Spec Reference :** 
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/media/ApplicationBasic.adoc

**XML Reference :** 
https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/application-basic-cluster.xml

bzbarsky-apple, 2023-11-09:
@Rajashreekalmane In what sense are global attributes missing in XML?  They are global; they are listed in the global attributes XML file, not with every cluster.

Rajashreekalmane, 2023-11-15:
I missunderstood the requirement, Thank you for the clarification related to global attributes.