project-chip/connectedhomeip #30357: [1.3] ICD Management Cluster XML has to be updated as per spec

Rajashreekalmane, 2023-11-09:
### Reproduction steps

**Description**
1. XML should remove the attributes which are not available in spec those are
UserActiveModeTriggerHint
UserActiveModeTriggerInstruction

2. below mentioned command is missed to define xml
StayActiveResponse

**Spec Reference :** 
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/ICDManagement.adoc

**XML Reference :** 
https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/icd-management-cluster.xml

### Bug prevalence

WSDFGHJ

### GitHub hash of the SDK that was being used

WERTHF

### Platform

raspi

### Type

Test Improvement

mkardous-silabs, 2023-11-17:
@Rajashreekalmane The attribute are in the spec but they are behind the in-progress field since they are aimed for 1.3.
The github does not show them; Check the code directly.

mkardous-silabs, 2023-11-17:
Reopened since the StayActiveRespone is missing.

PR #30559 (2023-11-19): [ICD] Update ICDM Cluster XML to latest spec
#### Description 
fixes #30357
closed 2023-11-19 by mkardous-silabs