project-chip/connectedhomeip #30359: [1.3] Color Control Cluster XML has to be updated as per spec

Ashwinigrl, 2023-11-09:
### Reproduction steps

- [ ]  **"EnhancedColorMode" attribute max range is wrong in XML. It should be 3 instead of 0xFF.**
 
[image]
[image]

- [ ]  **Bit 0 Feature name has to be updated from “HueAndSaturation” to “Hue or Saturation” in XML.**
 
[image]
[image]

- [ ]  **"CoupleColorTempToLevelMinMireds" attribute Max range value is wrong in XML.**
[image]
[image]

- [ ]  **"EnhancedColorMode" and "DriftCompensation" Attributes eum values are missing in XML.**
[image]

- [ ]  **"Options" attribute bit value is missing in XML.**
[image]

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

f4735790763692462c0803be35c808d52741ffb4

### Platform

raspi

### Platform Version(s)

1.3

### Type

Test Improvement

### Anything else?

**Spec Reference:** https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/ColorControl.adoc

**XML Reference:** https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/color-control-cluster.xml

bzbarsky-apple, 2023-11-09:
This can't be done until https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/6174 and https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/6251 are fixed.

tcarmelveilleux, 2024-07-03:
"HueAndSaturation" is correct. That's the mode, which uses BOTH values at the same time. The `/` doesn't mean "Or", and there is no value making it an "Or" now. If anything, it changes the meaning from the spec and common understanding of color models