project-chip/connectedhomeip #30361: [1.3] Flow Measurement Cluster XML has to be updated as per spec

Ashwinigrl, 2023-11-09:
### Reproduction steps

- [ ] **ClusterRevision value has to be added in XML. The value is 3.** 

### XML Screenshot:
[image]
### Spec Screenshot:
[image]

- [ ]  **MeasuredValue attribute default value is missing in XML.**

### Spec Screenshot:
[image]
### XML Screenshot:
[image]

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

f4735790763692462c0803be35c808d52741ffb4

### Platform

raspi

### Platform Version(s)

1.3

### Type

Test Improvement

### Anything else?

**Spec Reference:** https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/FlowMeasurement.adoc

**XML Reference:** https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/flow-measurement-cluster.xml

bzbarsky-apple, 2023-11-09:
@Ashwinigrl Default values in XML have nothing to do with the Default column in the spec.  They don't need to be set to anything based on the spec.

bzbarsky-apple, 2023-11-09:
@Ashwinigrl Similarly, descriptions in XML and descriptions in the spec are not really related.