project-chip/connectedhomeip #30367: [1.3] Boolean State Cluster XML has to be updated as per spec

Saravana-kr22, 2023-11-09:
### Reproduction steps

As per the spec the StateChange event is optional , Hence in the XML file this should have optional="true" , But it is missing in the XML.

**Spec reference:** 

[image]

**XML Reference:** https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/boolean-state-cluster.xml

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

eb2f8f6ca95c181577415bc4b3e7cc240d5b2edd

### Platform

raspi

### Platform Version(s)

RPI - 4, 8GB RAM

### Type

Test Improvement

bzbarsky-apple, 2023-11-09:
@Saravana-kr22 See https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/6270

ReneJosefsen, 2024-01-18:
The water management effort is a "consumer" of the Boolean State cluster, but I'm not sure this should be marked in the Water Management project 🤔

ReneJosefsen, 2024-01-18:
A spec PR has been raised to make it clear the the event is optional, so I assume we need to get optional="true" added to the XML.