project-chip/connectedhomeip #30371: [1.3] Pressure Measurement Cluster  XML has to be updated as per spec

Rajashreekalmane, 2023-11-09:
### Reproduction steps

**Description :**
Constraints are not defined for MeasuredValue, MinMeasuredValue, MaxMeasuredValue, Scale, ScaledValue, MinScaledValue and MaxScaledValue in XML.

[image]
[image]

**PFA Spec Reference :** 
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/PressureMeasurement.adoc#attributes

**PFA XML Reference :** 
https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/pressure-measurement-cluster.xml

### Bug prevalence

123

### GitHub hash of the SDK that was being used

34354E

### Platform

raspi

### Type

Test Improvement

bzbarsky-apple, 2023-11-09:
Constraints for readonly attributes are ignored in ZAP XML.

bzbarsky-apple, 2023-11-09:
And non-constant constraints are not expressible anyway.

closed 2023-11-09 by bzbarsky-apple