project-chip/connectedhomeip #30374: [1.3] Target Navigator Cluster

Rajashreekalmane, 2023-11-09:
### Reproduction steps

1. XML missed to define the CurrentTarget as ""Nullable"" and need to remove the default value as 0 (default value doesn't defined in Spec) for Current target attribute.

**PFA Spec screenshot :** 
[image]

**PFA XML screenshot :** 
[image]

2. Below mentioned event missed to define in XML file 
    TargetUpdated

[image]

### Bug prevalence

sdfgthuj

### GitHub hash of the SDK that was being used

aSdzfgthj

### Platform

raspi

### Type

Test Improvement

bzbarsky-apple, 2023-11-09:
@Rajashreekalmane default values in XML have nothing to do with the Default column in the spec.

The CurrentTarget thing needs to be changed in the spec; changing it in the XML is a breaking change.  Please file a spec issue.

Rajashreekalmane, 2023-11-10:
Additionally,  cluster revision attribute should be added to the XML file, specifying the value as 2
[image]