project-chip/connectedhomeip #30383: [1.3] Laundry Washer Mode Cluster XML has to be updated as per spec

Ashwinigrl, 2023-11-09:
### Reproduction steps

- [ ] **"StartUpMode" and "OnMode" attributes are marked as provisional in spec, So it should be removed from XML.**

**Screenshot:**
[image]

[image]
- [ ]  **"SupportedModes" atribute min range is missing in XML**

**Screenshot:**
[image]

[image]

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

f4735790763692462c0803be35c808d52741ffb4

### Platform

raspi

### Platform Version(s)

1.3

### Type

Test Improvement

### Anything else?

**Spec Reference:**

- https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/Mode_Laundry.adoc#5-attributes

- https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/ModeBase.adoc#6-attributes

**XML Reference:**
- https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/laundry-washer-mode-cluster.xml

bzbarsky-apple, 2023-11-09:
Things cannot be removed from XML once they ship in client APIs. Has this shipped in client APIs?

bzbarsky-apple, 2024-12-10:
These are now X in the spec but still present in the SDK XML for some reason....