project-chip/connectedhomeip #30401: [Scenes] Removing mNameLenght form SceneData in SceneTableImpl

lpbeliveau-silabs, 2023-11-09:
We need to remove mNameLength and ensure trailing null character removal.

Also, if https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/8022 is accepted, we can move to determine end of string with null terminators.

lpbeliveau-silabs, 2023-11-09:
See https://github.com/project-chip/connectedhomeip/pull/30393#discussion_r1388596042 (and the remainder of the PR)

bzbarsky-apple, 2023-11-10:
The issue is not "trailing null character" so much as "you should be returning the strings you actually have, not something that includes extra garbage", no?

And no, you cannot move to determining end of string "with null terminators" for a data model string (i.e. CharSpan) someone hands you.

andy31415, 2023-11-10:
Technically including a null inside a string is allowed by the spec (you can have inside null, just no null at the end ... fun... I complained in the spec about this).

lpbeliveau-silabs, 2023-12-12:
Given the comments above, I am closing the issue.

closed 2023-12-12 by lpbeliveau-silabs