project-chip/connectedhomeip #30433: Compilation getting stuck [1.2]

Abhishek904094, 2023-11-13:
### Reproduction steps

Try to build version 1.2.0.0 after giving git Xdf clean

### Bug prevalence

[0/172] c++ obj/commands/discover/chip-tool-utils.DiscoverCommand.cpp while compiling 1.2v it is getting stuck at his file every time and not proceeding forward .Rpi is getting hanged every time need to reboot still it is not proceeding a single step from this file

### GitHub hash of the SDK that was being used

Dont know

### Platform

raspi

### Platform Version(s)

4

### Type

Platform Issue

### Anything else?

Rpi getting need to fix to resolve this issue

jacqueshsu, 2023-11-13:
To use -j1 try to it.

Abhishek904094, 2023-11-14:
To use -j1 try to it  still not working getting stuck at same file 
[0/172] c++ obj/commands/discover/chip-tool-utils.DiscoverCommand.cpp.

jay-manikandan, 2023-11-20:
I too had faced the similar issue. Did git pull and submodule init. If your build is getting stuck, try -j1. I too had tried with -j1 when my build got stuck.

git pull
git submodule update --init

gn gen out/host
ninja -j1 -C out/host