project-chip/connectedhomeip #30472: [BUG] The command handler state tracking does not work properly for the groupcast messages.

kkasperczyk-no, 2023-11-14:
### Reproduction steps

The `DispatchCommand` is called the same way for both groupcast and unicast scenarios: https://github.com/project-chip/connectedhomeip/blob/master/src/app/CommandHandler.cpp#L435 and it doesn't pass the information about the originator of this call. It results in attempts of updating the status (with `AddStatus` method) for the invoking response purposes.

For the groupcast scenario updating status does not make sense, as response will not be sent anyway and additionally it leads to the application being killed if more than one endpoint existing on a single Matter node was added to the same group.
In such case receiving the groupcast command invokes the status update, which works well for the first endpoint from a list, but it fails for the following ones on a following check: https://github.com/project-chip/connectedhomeip/blob/master/src/app/CommandHandler.cpp#L549

This issue affects all devices that would like to configure single group on more than one endpoint within the single Matter node and also bridge devices that may represent several different bridged devices belonging to a single group.

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

Tested with 181b0cb14ff007ec912f2ba6627e05dfb066c008, but it applies to master as well.

### Platform

core

### Platform Version(s)

All

tehampson, 2023-11-14:
I think [the suggestion that was proposed in slack](https://csamembers.slack.com/archives/G014G30SVV0/p1699976731844579?thread_ts=1699971396.191729&cid=G014G30SVV0) is instead of plumbing a flag through `CommandHandler::DispatchCommand` all the way back to `CommandHandler::AddStatus`. That you just indicate through some state variable in `CommandHandler` that we are handling a group command and that way it gets dropped once `CommandHandler::AddStatus` is called

kkasperczyk-no, 2023-11-15:
TODO: Add yaml tests covering the scenario of using multiple endpoints belonging to the same group and existing on a single Matter node.

PR #30484 (2023-11-15): Fix handling command handler status for group messages
The command handler status is added for the response purposes even in case of invoking commands targeted to the group. In case of using more than one endpoint on a single Matter node that was added to the same group, it leads to the application crash. The direct reason is a state check, that succeeds only for the first endpoint and fails for the subsequent calls.

Added a new state flag, which informs if the command that is currently handled origins from groupcast communication or not. In case of targeting the group, application returns pr [... cut]
commit 1ada682e (2023-11-15): Fix handling command handler status for group messages (#30484)

The command handler status is added for the response purposes
even in case of invoking commands targeted to the group.
In case of using more than one endpoint on a single
Matter node that was added to the same group, it leads to the
application crash. The direct reason is a state check,
that succeeds only for the first endpoint and fails for the
subsequent calls.

Added a new state flag, which informs if the command that is
currently handled origins from groupcast communication or not.
In case of targeting the group, application  [... cut]
closed 2024-01-04 by mergify[bot]