project-chip/connectedhomeip #30488: [1.3] RVC Clean Mode  Cluster XML has to be updated as per spec

Ashwinigrl, 2023-11-15:
### Reproduction steps

- [ ]  **"SupportedModes" atribute min range is missing in XML**

**Screenshot:**

[image]
[image]

- [ ]  **"OnMode"  is `OnOff` Feature dependent attribute. This feature is marked as X in conformance. (X means -   This is disallowed for the cluster derivation),  but it is reflecting in XML file.**

**Screenshot:**

[image]

[image]

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

f4735790763692462c0803be35c808d52741ffb4

### Platform

raspi

### Platform Version(s)

1.3

### Type

Test Improvement

### Anything else?

**Spec Reference:**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/device_types/RoboticVacuumCleaner.adoc#element-requirements

**XML Reference:**
https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/rvc-clean-mode-cluster.xml

hicklin, 2024-01-16:
The required change in the spec to address this issue was made in [this PR](https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/8637).

The required changes to the SDK following this spec change PR are tracked by [this issue](https://github.com/project-chip/connectedhomeip/issues/31447).

hicklin, 2024-01-30:
This issue was addressed by PR #31451.

closed 2024-01-30 by hicklin