project-chip/connectedhomeip #30489: [1.3] RVC RUN Mode Cluster XML has to be updated as per spec

Ashwinigrl, 2023-11-15:
### Reproduction steps

### Description:

- [ ]  **"OnMode"  is `OnOff` Feature dependent attribute. This feature is marked as X in conformance. (X means -   This is disallowed for the cluster derivation),  but it is reflecting in XML file.**

**Screenshot:**

[image]

[image]

**Spec Reference:**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/device_types/RoboticVacuumCleaner.adoc#element-requirements

**XML Reference:**
https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/rvc-run-mode-cluster.xml

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

f4735790763692462c0803be35c808d52741ffb4

### Platform

raspi

### Platform Version(s)

1.3

### Type

Test Improvement

bzbarsky-apple, 2023-11-17:
@Ashwinigrl The text above does not match the screenshot.... Which is correct?

hicklin, 2024-01-16:
The required change in the spec to address this issue was made in [this PR](https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/8637).

The required changes to the SDK following this spec change PR are tracked by [this issue](https://github.com/project-chip/connectedhomeip/issues/31447).