project-chip/connectedhomeip #30499: [BUG] Darwin Matter extension not calling back to `commissionDevice` with payload

maurycyw, 2023-11-15:
### Reproduction steps

When using our 1st party matter device, the matter extension callback for `commissionDevice` is never being called. However when testing with a 3rd party tappo matter plug, we are finding that the `commissionDevice` callback is being received. Through debugging we discovered that the matter support framework actually commissions a device (to the apple fabric?) then opens a commissioning window. (I am hoping my understanding here is correct).

**Question**:
Is there any specific cluster implementation that is needed that we could be missing with our 1t party device? We recently tested and verified the need for the Administration cluster.  Any other reasons that `commissionDevice` callback is never invoked in the matter extension? We are getting past the wifi setup just not seeing `commissionDevice` being called.

Any help or guidance would be greatly appreciated.

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

8f66f4215bc0708efc8cc73bda80620e67d8955f

### Platform

darwin

bzbarsky-apple, 2023-11-17:
@maurycyw Please submit a Feedback Assistant ticket with the relevant logs and put the ticket id here.  I'll follow up on it.

I am not aware of anything in particular that your device needs to be doing (other than being commissionable and implementing the spec-required OpenCommissioningWindow command) to work here: MatterSupport commissions it into the keychain fabric and then opens a commissioning window, as you note, then hands the onboarding payload for that new commissioning window to the app that kicked off the process.

Feedback assistant ticket:
- File a feedback assistant ticket.
https://developer.apple.com/bug-reporting/#feedback-assistant
- Click on the link below using your account:
https://feedbackassistant.apple.com/new-form-response
- Choose iOS/iPadOS to file an issue
- Choose the area as Home App and fill in the details of the type of issue and the description
* Technical Information
Please provide the following
a) Sysdiagnose logs for iPhone / iPad with the timestamp / timeframe of the issue
b) Sysdiagnose logs for all the Home Hubs (HomePod mini / HomePod / Apple TV) with the timestamp / timeframe of the issue
c) Accessory logs with the timestamp / timeframe of the issue
d) mDNS network logs with timestamp / timeframe of the issue
e) Video or screenshot describing the issue (optional but beneficial)
Please make sure you download and install the profiles for iOS and tvOS as provided here. The instructions are also in these links.
* iOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=homekit
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=network
 - https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=ios&name=mdns
* tvOS:
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=homekit
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=network
- https://developer.apple.com/bug-reporting/profiles-and-logs/?platform=tvos&name=mdns
For installation instructions, refer to https://developer.apple.com/bug-reporting/profiles-and-logs.
Note: Once you have the profile on the iPhone, it will ask you if you want to install it on other devices. Follow the prompts to install on the HomePod, Apple TV, etc., where you have the described issue and reproduce it.

huchuankan, 2024-03-14:
Matter Commission  demo video ： https://www.bilibili.com/video/BV1Nx421y7sa/

maurycyw, 2024-04-30:
@bzbarsky-apple 
Seems like a device must report a (non root) endpoint with device types and otherwise the extension silently fails. After making this change the callback succeeded.

bzbarsky-apple, 2024-04-30:
@maurycyw Thank you for figuring that out!

aajain-com, 2024-05-02:
@maurycyw  would you be able to please file an Apple Feedback request based on the instructions above and share the feedback ticket number here?