project-chip/connectedhomeip #30500: [TC-SC-5.3] Question about the DISZ flag in test case TC-SC-5.3 step 6

lion6420, 2023-11-16:
### Test issue(s)

In the test plan indicate that if group command is sent. The DSIZ flag is set to group.
The expected log, which is described in TC-SC-5.3, shows below
```
Verify the DSIZ flag is set to Group On TH(Reference app) log:
          (Here in below log the value recived is 0X04 = 0100 (decimal) - Here the bit 2 is enabled then DSIZ flag is set to group)

          [1686317564.742205][7679:7679] CHIP:DMG: Header Flags =
          [1686317564.742207][7679:7679] CHIP:DMG: {
          [1686317564.742212][7679:7679] CHIP:DMG:     Message (0x04) =
          [1686317564.742214][7679:7679] CHIP:DMG:     {
          [1686317564.742217][7679:7679] CHIP:DMG:         SourceNodeId = C831420E9B16886B
          [1686317564.742219][7679:7679] CHIP:DMG:     }
          [1686317564.742224][7679:7679] CHIP:DMG:     Exchange (0x03) =
          [1686317564.742226][7679:7679] CHIP:DMG:     {
          [1686317564.742228][7679:7679] CHIP:DMG:         Initiator = true
          [1686317564.742231][7679:7679] CHIP:DMG:         AckMsg = 48718950
          [1686317564.742233][7679:7679] CHIP:DMG:     }
          [1686317564.742237][7679:7679] CHIP:DMG: }
```

According to the spec,
If the S flag is set (Source Node ID field is present) and the DSIZ is set to 2 (Destination Node ID field presenting as a 16-bit Group ID) . The bits should be 0110 (0x06).
But the expected log in TC-SC-5.3 is 0x04.

Am I misunderstanding or missing something? 

### Platform

raspi

bzbarsky-apple, 2023-11-17:
@lion6420 This sounds like a bug in certification verification steps, yes?  Please file those in https://github.com/project-chip/matter-test-scripts

lion6420, 2023-11-17:
> This sounds like a bug in certification verification steps, yes?
@bzbarsky-apple Yes, I will change the repository.
Thanks for your reply!

closed 2023-11-17 by lion6420