project-chip/connectedhomeip #30505: [1.3] OpenIOT Builds failing quite a lot

woody-apple, 2023-11-16:


woody-apple, 2023-11-16:
```
2023-11-15T19:31:52.2102562Z 19:31:52.209   INFO   <--|FVPdev| [ TestClientMonitoringRegistrationTable ]
2023-11-15T19:31:52.2121111Z 19:31:52.211   INFO   <--|FVPdev| [ TestClientMonitoringRegistrationTable : Setup                                           ] : PASSED
2023-11-15T19:31:52.2227037Z 19:31:52.222   INFO   <--|FVPdev| [ TestClientMonitoringRegistrationTable : TestEntryKeyFunctions                           ] : PASSED
2023-11-15T19:31:52.2290807Z 19:31:52.228   INFO   <--|FVPdev| src/app/tests/TestICDMonitoringTable.cpp:94: assertion failed: "entry.IsKeyEquivalent(ByteSpan(entry2.key.As<Crypto::Aes128KeyByteArray>()))"
2023-11-15T19:31:52.2307510Z 19:31:52.230   INFO   <--|FVPdev| [ TestClientMonitoringRegistrationTable : TestEntryAssignationOverload                    ] : FAILED
2023-11-15T19:31:52.2498762Z 19:31:52.249   INFO   <--|FVPdev| [ TestClientMonitoringRegistrationTable : TestSaveAndLoadRegistrationValue                ] : PASSED
2023-11-15T19:31:52.2537606Z 19:31:52.253   INFO   <--|FVPdev| [ TestClientMonitoringRegistrationTable : TestSaveAllInvalidRegistrationValues            ] : PASSED
2023-11-15T19:31:52.2684760Z 19:31:52.268   INFO   <--|FVPdev| [ TestClientMonitoringRegistrationTable : TestSaveLoadRegistrationValueForMultipleFabrics ] : PASSED
2023-11-15T19:31:52.2904548Z 19:31:52.290   INFO   <--|FVPdev| [ TestClientMonitoringRegistrationTable : TestDeleteValidEntryFromStorage                 ] : PASSED
2023-11-15T19:31:52.2910842Z 19:31:52.290   INFO   <--|FVPdev| Failed Tests:   1 / 6
```

woody-apple, 2023-11-16:
Looks like this PR broke it:
https://github.com/project-chip/connectedhomeip/pull/30419

woody-apple, 2023-11-16:
Confirmed in the PR, it was the first instance of this failure, and the PR itself failed, but was merged.

jepenven-silabs, 2023-11-16:
My bad, CI was failling for another reason so I suspected a fluke, turns out that a real issue was hidden underneath the fluke. will fix asap

PR #30513 (2023-11-16): [CI] Fix OpenIoT failure for ICDMonitoringTableTest
Fix #30505 

PR #30385 introduced an error that broke OpenIoT Unit tests. This is the fix for said error.

OpenIoTSDK is the ONLY CI job that run Unit Test with the PSA Crypto implementation. With such implementation, Keys are not stored in plain text within a KeyHandle, they are stored somewhere else and only a KeyId is stored inside a KeyHandle. This why the test was succeeding for every other test but not OpenIoT...
closed 2023-11-16 by andy31415